jeanmarc77/123solar #1: Feature request. Add notification when there is no current on the array.

hmakmur, 2021-07-02:
After loosing over 10 months of solar production because fuses on my array was blown last year,I installed 123solar.   I was hoping that 123solar would help notify me when something bad happens to my solar array. Unfortunately this was not so. 123Solar helped make me aware  by luck today that my solar production is much lower than expected. Looking back at the graph, I notice 6 weeks ago one of the array shows 0 current on one of the array. 

Would it be possible to add a notification when an array went to 0 amp?  this would help tell when the fuses went bad. thanks!!

jeanmarc77, 2021-07-02:
I believe this sort of warning should come from your inverter protocol via alarms/warnings check. Which one are you using ?

hmakmur, 2021-07-02:
I use aurora protocol and it does not show any notification.  I looked the event file and  I got nothing. It would be nice to report 0 current if this is not too hard to do and it would work for any inverter.

jeanmarc77, 2021-07-03:
It could be done by tweaking aurora_checks.php to return $ALARM or $MESSAGE while checking $IxP.
You may have false warnings in case of snow covering panels and also at sunrise, sunset

hmakmur, 2021-07-03:
You are correct. I did  not think of the false alarm issues.  Alternatively,  I can connect 123Solar with HomeAssistant and do the notification logic there. Thanks for your quick response.