jeanmarc77/123solar #4: Feature request: is possible to have an API for todays production

lorenzo93, 2022-01-03:
Hi,

I use 123solar from 2014 and I'm very happy with it.
I'm trying to integrate it with OpenHab (my current domotic solution) and I've managed to integrate quite everything so far.

I'm just missing the todays production value. Unfortunatelly it is transmitted to the frontend as string (in `programday.php` as title) and I was not able to find another endpoint for the value.

Side question: I'm currently running version 1.5.5. Do you know any upgrade issues moving from that version to the current?

jeanmarc77, 2022-01-03:
Hi,
1.5.5 is more than 6 years old.. You need to convert data since 1.6 with https://github.com/jeanmarc77/123solar/blob/main/other/convert_to_1.6.tar.gz
I will release a new version on github that allow updating with github

You can have latest value with /dev/shm/123s_MEMORY.json "Last1" - "First1"

closed 2022-01-03 by jeanmarc77

lorenzo93, 2022-01-04:
Hi @jeanmarc77 

now I remember why I didn't upgrade to recent versions :)

Anyhow, I was looking for a `REST` endpoint for the today's production value, non a local file. The two systems are far from each-other and can talk just with `HTTP`.