jeanmarc77/123solar #7: 123solar go OFF himself

cerebrum2020, 2022-02-16:
Hi JeanMarc, 

I find a problem with buster, updated to january 2022.
With last version of 123solar (but also with old ones, other users reported same issue) 123solar go OFF exactly 1 hour after his morning awake.

After that:
- if I put manually ON, all work as before;
- if I reboot the the raspberry pi (3b+ in my case) all work as before, I find 123 solar ON and no problem till the next day

nothing on debug or nothing that have problem in the system, also meterN work properly.
Thanks for some support

jeanmarc77, 2022-02-16:
Hi,
Have you check your system date and timezone ? Your localizationis ok into 123s ?
Which protocol are you using ?
I believe your com app might been stuck. You can look if there is still an old /var/lock/LCK..thing file while it happen.

cerebrum2020, 2022-02-17:
Hi JeanMarc,

note that 123solar go in OFF in the admin page, the switch turn in red and  OFF position, everytime after 1 hour from the awake in the morming.
About your requests:

_- Have you check your system date and timezone ?_ Yes, it's ok
_- Your localization is ok into 123s ?_ Also, is it correct
_- Which protocol are you using ?_ sdm120, the old system ryn from 3 year without problems. Just after passing to buster I've this trouble
- _I believe your com app might been stuck. You can look if there is still an old /var/lock/LCK..thing file while it happen._ Also checked this morning, just the LCK..ttyUSB0 that is present before and after the problem.

This is the info page of 123 solar AFTER the reboot, at 8:16 more or less.
I find the switch in admin page turned ON and green. Imself.

```
#* 17/02/2022 08:17:33	123Solar awake

#* 17/02/2022 08:17:31	Starting 123Solar on boot debug (656)

#1 17/02/2022 07:14:59	Clean up log

#* 17/02/2022 07:14:01	123Solar awake

#* 16/02/2022 17:56:00	123Solar asleep
```

jeanmarc77, 2022-02-17:
Hi,
If it's one hour after awake, it might be something wrong while requesting sdm120c-pool_startup.php but that's really odd to me..

I have previsously had that issue using SDM120C v1.3.5.5, did you run v1.4 (Beta) from https://github.com/The-Drake/SDM120C ? That did the trick for me.

Also, if it continue to happen and if LCK is stuck, you could try this tools : https://github.com/jeanmarc77/meterN_comapps/blob/main/tools/clearsdm.php
Call it every 5min or so to clear the hanging process.

cerebrum2020, 2022-02-18:
Hi, 

just checked the SDM120C verions. Think is the v1.4 because I read 2 meters (in the note of the v1.4 this is a new feature, no t possibile with 1.3.x). But I've asked to Flanesi a confirm, the solarbuster image that I've used to start with my systestm is from him.

I'll try with che clearsdm and check if the issue it solved.
I'll update you, thanks.

jeanmarc77, 2022-02-18:
You can see sdm version by typing 'sdm120c'. Did you use sdm120c -z3 option ?

cerebrum2020, 2022-02-18:
Hi, checked at home
is the sdm120c 1.3.5.4 !
yes, used -z3 (-b 9600 -P N -z3 -w10 -j10)

jeanmarc77, 2022-02-24:
BTW you don't report error log but history.
To get debug log go to admin/help.php

cerebrum2020, 2022-03-01:
> BTW you don't report error log but history. To get debug log go to admin/help.php

The reported in previous post is the status, right.

But the php warning message is from the debug log ( I've activated from the beginning the debug to check if somthing goes wrong...).

But there is just starting 123solar in boot debug (id), day by day

jeanmarc77, 2022-03-05:
Which version of 123s are you using ?

cerebrum2020, 2022-03-05:
the last one
1.8.4.4
sdm120c

Tryed with a new anf fresh installation of the system with the same image used for the actual installation that have this problem  (started from buster, updated 123 solar and meterN). 
In 4 days no problem, but running in another raspberry not connected to the sdm120 meters (123solar ON and meterN ON).
Today I will finish the configuration and test the new installation paired with the meters....
Hoper this could solve.

jeanmarc77, 2022-03-05:
hard to tell without any debug log.
tell me how it went

jeanmarc77, 2022-04-01:
I guess i could close this as it's not an issue with 123s

cerebrum2020, 2022-04-01:
Hi JeanMarc 

you are right, sorry for my big delay in reply.
This week I've just re-installed all the system and re-configured every part.
The problem is preset again, with the new "hour" also....
Reboot and all goes ok...
In your opinion is not an 123solar problem?
Other to check??

thanks

jeanmarc77, 2022-04-01:
It's an not issue with 123s but maybe in protocol file.
If it happen, try to run "ps -ef | grep sdm120c" several times and "ls -al /var/lock/", to see if the process run and the last time it ran.

cerebrum2020, 2022-04-03:
Hi Jeanmarc
This is the screen when 123solar goes off, after 1 hour from awake (meantime metern n still running and work)
______________

_$ ps -ef | grep sdm120c
www-data 23196  1100  0 08:15 ?        00:00:00 sdm120c -a 6 -b 9600 -z 10 -i -p -v -c -f -g -P N -w 10 -j 10 -d 0 -q /dev/ttyUSB0
pi       23198 20405  0 08:15 pts/0    00:00:00 grep --color=auto sdm120c_

or 

_www-data 23196  1100  0 08:15 ?        00:00:00 sdm120c -a 6 -b 9600 -z 10 -i -p -v -c -f -g -P N -w 10 -j 10 -d 0 -q /dev/ttyUSB0_

(I've 2 phisical meters in consumption)
and

_$ ls -al /var/lock/
totale 8
drwxrwxrwt  4 root     root     120 apr  3 08:16 .
drwxr-xr-x 29 root     root     940 apr  3 08:15 ..
drwxr-xr-x  2 www-data root      40 apr  2 08:50 apache2
-rw-r--r--  1 root     root      22 apr  2 08:50 asound.state.lock
-rw-r--r--  1 www-data www-data  14 apr  3 08:16 LCK..ttyUSB0
drwxr-xr-x  2 root     root      40 apr  2 08:50 subsys_

________________

after reboot 

_$ ps -ef | grep sdm120c
www-data 20920  1096  0 08:23 ?        00:00:00 sdm120c -a 7 -b 9600 -z 10 -i -p -v -c -f -g -P N -w 10 -j 10 -d 0 -q /dev/ttyUSB0
www-data 20927   679  0 08:23 ?        00:00:00 sh -c sdm120c -a 1 -b 9600 -P N -z3 -w10 -j10 -vpcfi -q /dev/ttyUSB0 2>&1
www-data 20928 20927  0 08:23 ?        00:00:00 sdm120c -a 1 -b 9600 -P N -z3 -w10 -j10 -vpcfi -q /dev/ttyUSB0
pi       20930 16521  0 08:23 pts/0    00:00:00 grep --color=auto sdm120c_

and

_$ ls -al /var/lock/
totale 8
drwxrwxrwt  4 root     root     120 apr  3 08:23 .
drwxr-xr-x 29 root     root     940 apr  3 08:22 ..
drwxr-xr-x  2 www-data root      40 apr  3 08:19 apache2
-rw-r--r--  1 root     root      22 apr  3 08:19 asound.state.lock
-rw-r--r--  1 www-data www-data  28 apr  3 08:23 LCK..ttyUSB0
drwxr-xr-x  2 root     root      40 apr  3 08:19 subsys_

jeanmarc77, 2022-04-03:
Hi,
So it has stuck right at 08:15 or 8:16 ? when did you run the ps command, few minutes later ?

cerebrum2020, 2022-04-03:
It goes awake --> 06:51:15	123Solar awake
It stop after 1 hour ( I see pvoutput also without update )
I run the command to check at 8:15
I restart the system at 8:19 more or less
and after the command again

jeanmarc77, 2022-04-03:
So it mean the sdm120c still running. I don't know how the sdm120c-pool's pooler work. 
Maybe check the content and time of cat /dev/shm/metern*.txt files when it happen.
Also rather give a try using plain sdm120c protocol.

cerebrum2020, 2022-04-03:
But why 123solar just after 1 hour is dead, the main button (from admin panel) goes off?
This looks like is related to the 123solar?

jeanmarc77, 2022-04-03:
Maybe a segfault from a requested command but you said nothing on debug log (the one on help page)
I can't debug remotely, i am just guessing

jeanmarc77, 2022-04-17:
So how it went using sdm120c protocol ?

cerebrum2020, 2022-04-19:
Hi JeanMarc.
No way, same issue.
After 1 hour.... red stop.
I don't know what to try.
Maybe a downgrade of 123solar could help to understand some point?

jeanmarc77, 2022-04-19:
No need to downgrade 123s.
Maybe try some reliability test using [comtester](https://github.com/jeanmarc77/meterN_comapps/blob/main/tools/comtester.php) 
It should be use from CLI, turn off 123s.
Set $command = sdm120c -a 7 -vpcfi -q /dev/ttyUSB0.
You can change the script to run for few hours and request both adresses (1 & 7)

cerebrum2020, 2022-04-21:
Ok, great
I'll try to get some infos and test in next days, and update.

thanks

cerebrum2020, 2022-05-13:
Hi Jeanmarc,

I'm made some test and try to collect Infos.
Fresh installation and male all by the start.
Sadely I can work at this only during weekend and this Need some times.
I'll come back with all the Infos that I can collect.
Till today the problema exist..

jeanmarc77, 2022-07-19:
not enough details, i close

closed 2022-07-19 by jeanmarc77