jeanmarc77/123solar #8: abbuno protocol - debug output filling

flanesi, 2022-03-10:
I use 123solar with the abbuno protocol for reading the inverter.

123solar works regularly, but if I enable debugging it fills up with abbuno readings, like these:

```
array(12) {
  [0]=>
  string(55) "modpoll 3.10 - FieldTalk(tm) Modbus(R) Master Simulator"
  [1]=>
  string(39) "Copyright (c) 2002-2021 proconX Pty Ltd"
  [2]=>
  string(66) "Visit https://www.modbusdriver.com for Modbus libraries and tools."
  [3]=>
  string(0) ""
  [4]=>
  string(39) "Protocol configuration: MODBUS/TCP, FC3"
  [5]=>
  string(77) "Slave configuration...: address = 1, start reference = 40094 (PDU), count = 1"
  [6]=>
  string(77) "Communication.........: 192.168.2.22, port 502, t/o 1.00 s, poll rate 1000 ms"
  [7]=>
  string(71) "Data type.............: 32-bit integer, output (holding) register table"
  [8]=>
  string(67) "Word swapping.........: Slave configured as big-endian word machine"
  [9]=>
  string(0) ""
  [10]=>
  string(19) "-- Polling slave..."
  [11]=>
  string(15) "[40094]: 114214"
}
DEBUGGING:
KWHT: 1142.14
I1V : 325.2 
I1A : 6.4 
I1P : 2070 
I2V : 0 
I2A : 0 
I2P : 0 
G1V : 242.6  
G2V :   
G3V :   
G1A : 8.2  
G2A :   
G3A :   
G1P : 2010  
G2P :   
G3P :   
FRQ : 50 
INVT: 26.9
BOOT: 32.9
EFF:  97.1 
array(37) {
  [0]=>
  string(55) "modpoll 3.10 - FieldTalk(tm) Modbus(R) Master Simulator"
  [1]=>
  string(39) "Copyright (c) 2002-2021 proconX Pty Ltd"
  [2]=>
  string(66) "Visit https://www.modbusdriver.com for Modbus libraries and tools."
  [3]=>
  string(0) ""
  [4]=>
  string(39) "Protocol configuration: MODBUS/TCP, FC3"
  [5]=>
  string(78) "Slave configuration...: address = 1, start reference = 41123 (PDU), count = 26"
  [6]=>
  string(77) "Communication.........: 192.168.2.22, port 502, t/o 1.00 s, poll rate 1000 ms"
  [7]=>
  string(72) "Data type.............: 16-bit register, output (holding) register table"
  [8]=>
  string(67) "Word swapping.........: Slave configured as big-endian word machine"
  [9]=>
  string(0) ""
  [10]=>
  string(19) "-- Polling slave..."
  [11]=>
  string(11) "[41123]: 64"
  [12]=>
  string(13) "[41124]: 3252"
  [13]=>
  string(12) "[41125]: 207"
  [14]=>
  string(10) "[41126]: 0"
  [15]=>
  string(10) "[41127]: 0"
  [16]=>
  string(11) "[41128]: -1"
  [17]=>
  string(11) "[41129]: -1"
  [18]=>
  string(15) "[41130]: -32768"
  [19]=>
  string(10) "[41131]: 4"
  [20]=>
  string(10) "[41132]: 0"
  [21]=>
  string(10) "[41133]: 0"
  [22]=>
  string(13) "[41134]: -306"
  [23]=>
  string(10) "[41135]: 1"
  [24]=>
  string(10) "[41136]: 0"
  [25]=>
  string(13) "[41137]: -304"
  [26]=>
  string(11) "[41138]: 20"
  [27]=>
  string(10) "[41139]: 0"
  [28]=>
  string(10) "[41140]: 0"
  [29]=>
  string(10) "[41141]: 0"
  [30]=>
  string(10) "[41142]: 0"
  [31]=>
  string(10) "[41143]: 0"
  [32]=>
  string(10) "[41144]: 0"
  [33]=>
  string(10) "[41145]: 0"
  [34]=>
  string(11) "[41146]: -1"
  [35]=>
  string(10) "[41147]: 0"
  [36]=>
  string(10) "[41148]: 0"
}
array(61) {
  [0]=>
  string(55) "modpoll 3.10 - FieldTalk(tm) Modbus(R) Master Simulator"
  [1]=>
  string(39) "Copyright (c) 2002-2021 proconX Pty Ltd"
  [2]=>
  string(66) "Visit https://www.modbusdriver.com for Modbus libraries and tools."
  [3]=>
  string(0) ""
  [4]=>
  string(39) "Protocol configuration: MODBUS/TCP, FC3"
  [5]=>
  string(78) "Slave configuration...: address = 1, start reference = 40072 (PDU), count = 50"
  [6]=>
  string(77) "Communication.........: 192.168.2.22, port 502, t/o 1.00 s, poll rate 1000 ms"
  [7]=>
  string(72) "Data type.............: 16-bit register, output (holding) register table"
  [8]=>
  string(67) "Word swapping.........: Slave configured as big-endian word machine"
  [9]=>
  string(0) ""
  [10]=>
  string(19) "-- Polling slave..."
  [11]=>
  string(11) "[40072]: 82"
  [12]=>
  string(11) "[40073]: 82"
  [13]=>
  string(11) "[40074]: -1"
  [14]=>
  string(11) "[40075]: -1"
  [15]=>
  string(11) "[40076]: -1"
  [16]=>
  string(11) "[40077]: -1"
  [17]=>
  string(11) "[40078]: -1"
  [18]=>
  string(11) "[40079]: -1"
  [19]=>
  string(13) "[40080]: 2426"
  [20]=>
  string(11) "[40081]: -1"
  [21]=>
  string(11) "[40082]: -1"
  [22]=>
  string(11) "[40083]: -1"
  [23]=>
  string(12) "[40084]: 201"
  [24]=>
  string(10) "[40085]: 1"
  [25]=>
  string(13) "[40086]: 5000"
  [26]=>
  string(11) "[40087]: -2"
  [27]=>
  string(12) "[40088]: 201"
  [28]=>
  string(10) "[40089]: 1"
  [29]=>
  string(10) "[40090]: 0"
  [30]=>
  string(10) "[40091]: 1"
  [31]=>
  string(14) "[40092]: -9999"
  [32]=>
  string(11) "[40093]: -4"
  [33]=>
  string(10) "[40094]: 1"
  [34]=>
  string(15) "[40095]: -16858"
  [35]=>
  string(10) "[40096]: 1"
  [36]=>
  string(11) "[40097]: 64"
  [37]=>
  string(11) "[40098]: -1"
  [38]=>
  string(11) "[40099]: -1"
  [39]=>
  string(15) "[40100]: -32768"
  [40]=>
  string(12) "[40101]: 207"
  [41]=>
  string(10) "[40102]: 1"
  [42]=>
  string(12) "[40103]: 269"
  [43]=>
  string(12) "[40104]: 354"
  [44]=>
  string(15) "[40105]: -32768"
  [45]=>
  string(12) "[40106]: 329"
  [46]=>
  string(11) "[40107]: -1"
  [47]=>
  string(10) "[40108]: 4"
  [48]=>
  string(10) "[40109]: 6"
  [49]=>
  string(10) "[40110]: 0"
  [50]=>
  string(10) "[40111]: 0"
  [51]=>
  string(10) "[40112]: 0"
  [52]=>
  string(10) "[40113]: 0"
  [53]=>
  string(10) "[40114]: 0"
  [54]=>
  string(10) "[40115]: 0"
  [55]=>
  string(10) "[40116]: 0"
  [56]=>
  string(10) "[40117]: 0"
  [57]=>
  string(10) "[40118]: 0"
  [58]=>
  string(10) "[40119]: 0"
  [59]=>
  string(10) "[40120]: 0"
  [60]=>
  string(10) "[40121]: 0"
}
```

jeanmarc77, 2022-03-10:
Hi,
I did not write this file, it's too verbose in debug (var_dump)
Can you try this as protocols/abbuno.php ?

```
<?php
/**
 * /srv/http/123solar/scripts/protocols/abbuno.php
 *
 * @package default
 */

if (!defined('checkaccess')) {
    die('Direct access not permitted');
}

// For TCP Modbus https://www.modbusdriver.com/modpoll.html

$RET = '';
unset($CMD_RETURN);

if ($DEBUG) {
    $CMD_POOLING = "modpoll -m tcp -r 41123 -c 26 -0 -a 1 -i -1 ${'ADR'.$invt_num} 2> /tmp/modpoll.err";
} else {
    $CMD_POOLING = "modpoll -m tcp -r 41123 -c 26 -0 -a 1 -i -1 ${'ADR'.$invt_num}";
}
exec($CMD_POOLING, $CMD_RETURN);

if (isset($CMD_RETURN[11])) {
    // String 1
    $I1V = ((float) (substr($CMD_RETURN[12], 9)) / 10);
    $I1A = ((float) (substr($CMD_RETURN[11], 9)) / 10);
    $I1P = ((float) (substr($CMD_RETURN[13], 9)) * 10);
    
    // String 2
    $I2V = ((float) (substr($CMD_RETURN[32], 9)) / 10);
    $I2A = ((float) (substr($CMD_RETURN[31], 9)) / 10);
    $I2P = ((float) (substr($CMD_RETURN[33], 9)) * 10);
    
    // Other strings are not present
    $I3V = null;
    $I3A = null;
    $I3P = null;
    $I4V = null;
    $I4A = null;
    $I4P = null;
    
    // 2nd command to get grid parameters
    unset($CMD_RETURN);
    if ($DEBUG) {
        $CMD_POOLING = "modpoll -m tcp -r 40072 -c 50 -0 -a 1 -i -1 ${'ADR'.$invt_num} 2> /tmp/modpoll.err";
    } else {
        $CMD_POOLING = "modpoll -m tcp -r 40072 -c 50 -0 -a 1 -i -1 ${'ADR'.$invt_num}";
    }
    exec($CMD_POOLING, $CMD_RETURN);
    // Grid
    if (!${'PHASE' . $invt_num}) {
        $G1V = ((float) (substr($CMD_RETURN[19], 9)) / 10);
        $G1A = ((float) (substr($CMD_RETURN[11], 9)) / 10);
        $G1P = ((float) (substr($CMD_RETURN[23], 9)) * 10);
        $G2V = null;
        $G2A = null;
        $G2P = null;
        $G3V = null;
        $G3A = null;
        $G3P = null;
    } else {
        $G1V = ((float) (substr($CMD_RETURN[19], 9)) / 10);
        $G1A = ((float) (substr($CMD_RETURN[12], 9)) / 10);
        $G1P = round(($G1V * $G1A), 3);
        $G2V = ((float) (substr($CMD_RETURN[20], 9)) / 10);
        $G2A = ((float) (substr($CMD_RETURN[13], 9)) / 10);
        $G2P = round(($G2V * $G2A), 3);
        $G3V = ((float) (substr($CMD_RETURN[21], 9)) / 10);
        $G3A = ((float) (substr($CMD_RETURN[14], 9)) / 10);
        $G3P = round(($G3V * $G3A), 3);
    }
    
    $FRQ  = ((float) (substr($CMD_RETURN[25], 9)) / 100);
    $INVT = ((float) (substr($CMD_RETURN[42], 9)) / 10);
    $BOOT = ((float) (substr($CMD_RETURN[45], 9)) / 10);
    $EFF = round((100.0 * (float) (substr($CMD_RETURN[23], 9)) / (float) (substr($CMD_RETURN[40], 9))), 2);
    
    if ($FRQ > 0 && $EFF > 0 && $EFF < 1.0) { // Avoid null values at early startup
        $RET = 'OK';
    } else {
        $RET = 'NOK';
    }
    // 3 command to get total energy
    unset($CMD_RETURN);
    if ($DEBUG) {
        $CMD_POOLING = "modpoll -m tcp -t4:int -r 40094 -c 1 -0 -a 1 -i -1 ${'ADR'.$invt_num} 2> /tmp/modpoll.err";
    } else {
        $CMD_POOLING = "modpoll -m tcp -t4:int -r 40094 -c 1 -0 -a 1 -i -1 ${'ADR'.$invt_num}";
        
    }
    exec($CMD_POOLING, $CMD_RETURN);
    $KWHT = ((float) (substr($CMD_RETURN[11], 9)) / 100);
    
    if ($KWHT > 0) { // Avoid null values due to communication error
        $RET = 'OK';
    } else {
        $RET = 'NOK';
    }
} else {
    $RET = 'NOK';
    if ($DEBUG) {
        $time = date('Ymd-H:i:s');
        exec('cp /tmp/modpoll.err ' . $INVTDIR . '/errors/modpoll' . $time . '.err');
        file_put_contents($INVTDIR . '/errors/out' . $time . '.txt', $CMD_RETURN);
    }
    $CMD_RETURN = 'Error on Modpoll';
}
?>
```

flanesi, 2022-03-10:
Now works.
Thank you

jeanmarc77, 2022-03-10:
Great, can you try this one ?
`<?php
/**
 * /srv/http/123solar/scripts/protocols/abbuno.php
 *
 * @package default
 */

if (!defined('checkaccess')) {
    die('Direct access not permitted');
}

// For TCP Modbus https://www.modbusdriver.com/modpoll.html

$RET = '';
$cmdout = ''; // 3 commands
unset($CMD_RETURN);

if (!$DEBUG) {
    $CMD_POOLING = "modpoll -m tcp -r 41123 -c 26 -0 -a 1 -i -1 ${'ADR'.$invt_num}";
	exec($CMD_POOLING, $CMD_RETURN);
} else {
    $CMD_POOLING = "modpoll -m tcp -r 41123 -c 26 -0 -a 1 -i -1 ${'ADR'.$invt_num} 2> /tmp/modpoll.err";
	exec($CMD_POOLING, $CMD_RETURN);
	$cmdout = implode(',', $CMD_RETURN);
}

if (isset($CMD_RETURN[13])) {
    // String 1
    $I1V = ((float) (substr($CMD_RETURN[12], 9)) / 10);
    $I1A = ((float) (substr($CMD_RETURN[11], 9)) / 10);
    $I1P = ((float) (substr($CMD_RETURN[13], 9)) * 10);
    // String 2
    $I2V = ((float) (substr($CMD_RETURN[32], 9)) / 10);
    $I2A = ((float) (substr($CMD_RETURN[31], 9)) / 10);
    $I2P = ((float) (substr($CMD_RETURN[33], 9)) * 10);
    // Other strings are not present
    $I3V = null;
    $I3A = null;
    $I3P = null;
    $I4V = null;
    $I4A = null;
    $I4P = null;
} else {
    $RET = 'NOK';
    $CMD_RETURN = 'Error on Modpoll: 1st command';
}

unset($CMD_RETURN);
// 2nd command to get grid parameters
if (!$DEBUG) {
	$CMD_POOLING = "modpoll -m tcp -r 40072 -c 50 -0 -a 1 -i -1 ${'ADR'.$invt_num}";
	exec($CMD_POOLING, $CMD_RETURN);
} else {
	$CMD_POOLING = "modpoll -m tcp -r 40072 -c 50 -0 -a 1 -i -1 ${'ADR'.$invt_num} 2>> /tmp/modpoll.err";
	exec($CMD_POOLING, $CMD_RETURN);
	$cmdout .= implode(',', $CMD_RETURN);
}
if (isset($CMD_RETURN[45])) {
    // Grid
    if (!${'PHASE' . $invt_num}) {
        $G1V = ((float) (substr($CMD_RETURN[19], 9)) / 10);
        $G1A = ((float) (substr($CMD_RETURN[11], 9)) / 10);
        $G1P = ((float) (substr($CMD_RETURN[23], 9)) * 10);
        $G2V = null;
        $G2A = null;
        $G2P = null;
        $G3V = null;
        $G3A = null;
        $G3P = null;
    } else {
        $G1V = ((float) (substr($CMD_RETURN[19], 9)) / 10);
        $G1A = ((float) (substr($CMD_RETURN[12], 9)) / 10);
        $G1P = round(($G1V * $G1A), 3);
        $G2V = ((float) (substr($CMD_RETURN[20], 9)) / 10);
        $G2A = ((float) (substr($CMD_RETURN[13], 9)) / 10);
        $G2P = round(($G2V * $G2A), 3);
        $G3V = ((float) (substr($CMD_RETURN[21], 9)) / 10);
        $G3A = ((float) (substr($CMD_RETURN[14], 9)) / 10);
        $G3P = round(($G3V * $G3A), 3);
    }
    
    $FRQ  = ((float) (substr($CMD_RETURN[25], 9)) / 100);
    $INVT = ((float) (substr($CMD_RETURN[42], 9)) / 10);
    $BOOT = ((float) (substr($CMD_RETURN[45], 9)) / 10);
    $EFF = round((100.0 * (float) (substr($CMD_RETURN[23], 9)) / (float) (substr($CMD_RETURN[40], 9))), 2);
    if ($FRQ <= 0) { // Avoid null values at early startup
        $RET = 'NOK';
    }
} else {
	$RET = 'NOK';
    $CMD_RETURN = 'Error on Modpoll: 2nd command';
}

unset($CMD_RETURN);
// 3 command to get total energy
if (!$DEBUG) {
	$CMD_POOLING = "modpoll -m tcp -t4:int -r 40094 -c 1 -0 -a 1 -i -1 ${'ADR'.$invt_num}";
	exec($CMD_POOLING, $CMD_RETURN);
} else {
	$CMD_POOLING = "modpoll -m tcp -t4:int -r 40094 -c 1 -0 -a 1 -i -1 ${'ADR'.$invt_num} 2>> /tmp/modpoll.err";
	exec($CMD_POOLING, $CMD_RETURN);
	$cmdout .= implode(',', $CMD_RETURN);
}
if (isset($CMD_RETURN[11])) {
    $KWHT = ((float) (substr($CMD_RETURN[11], 9)) / 100);
    if ($KWHT <= 0) { // Avoid null values due to communication error
        $RET = 'NOK';
    }
} else {
    $RET = 'NOK';
    $CMD_RETURN = 'Error on Modpoll: 3th command';
}

if ($RET != 'NOK' && !empty($RET)) {
	$RET = 'OK';
} else {
	if ($DEBUG) {
        $time = date('Ymd-H:i:s');
        exec('cp /tmp/modpoll.err ' . $INVTDIR . '/errors/modpoll' . $time . '.err');
        file_put_contents($INVTDIR . '/errors/out' . $time . '.txt', $cmdout);
	}
}

?>`

flanesi, 2022-03-10:
This does not work.

Test from admin interface;
`modpoll -m tcp -t4:int -r 40094 -c 1 -0 -a 1 -i -1 192.168.2.22 return : Array`

Error in 123solar.err :
`PHP Warning:  trim() expects parameter 1 to be string, array given in /var/www/123solar/scripts/123solar.php on line 506`

jeanmarc77, 2022-03-10:
Maybe this ?
[test.txt](https://github.com/jeanmarc77/123solar/files/8225129/test.txt)

flanesi, 2022-03-10:
unfortunately it doesn't work..

Test from admin interface
```
modpoll -m tcp -r 40072 -c 50 -0 -a 1 -i -1 192.168.2.22 return : modpoll 3.10 - FieldTalk(tm) Modbus(R) Master Simulator,Copyright (c) 2002-2021 proconX Pty Ltd,Visit https://www.modbusdriver.com for Modbus libraries and tools.,,Protocol configuration: MODBUS/TCP, FC3,Slave configuration...: address = 1, start reference = 41123 (PDU), count = 26,Communication.........: 192.168.2.22, port 502, t/o 1.00 s, poll rate 1000 ms,Data type.............: 16-bit register, output (holding) register table,Word swapping.........: Slave configured as big-endian word machine,,-- Polling slave...,[41123]: 0,[41124]: 3012,[41125]: 2,[41126]: 0,[41127]: 0,[41128]: -1,[41129]: -1,[41130]: -32768,[41131]: 4,[41132]: 0,[41133]: 0,[41134]: -306,[41135]: 1,[41136]: 0,[41137]: -304,[41138]: 20,[41139]: 0,[41140]: 0,[41141]: 0,[41142]: 0,[41143]: 0,[41144]: 0,[41145]: 0,[41146]: -1,[41147]: 0,[41148]: 0,modpoll 3.10 - FieldTalk(tm) Modbus(R) Master Simulator,Copyright (c) 2002-2021 proconX Pty Ltd,Visit https://www.modbusdriver.com for Modbus libraries and tools.,,Protocol configuration: MODBUS/TCP, FC3,Slave configuration...: address = 1, start reference = 40072 (PDU), count = 50,Communication.........: 192.168.2.22, port 502, t/o 1.00 s, poll rate 1000 ms,Data type.............: 16-bit register, output (holding) register table,Word swapping.........: Slave configured as big-endian word machine,,-- Polling slave...,[40072]: 1,[40073]: 1,[40074]: -1,[40075]: -1,[40076]: -1,[40077]: -1,[40078]: -1,[40079]: -1,[40080]: 2402,[40081]: -1,[40082]: -1,[40083]: -1,[40084]: 0,[40085]: 1,[40086]: 5002,[40087]: -2,[40088]: 0,[40089]: 1,[40090]: 0,[40091]: 1,[40092]: 9979,[40093]: -4,[40094]: 1,[40095]: -15943,[40096]: 1,[40097]: 0,[40098]: -1,[40099]: -1,[40100]: -32768,[40101]: 2,[40102]: 1,[40103]: 226,[40104]: 196,[40105]: -32768,[40106]: 176,[40107]: -1,[40108]: 4,[40109]: 6,[40110]: 0,[40111]: 0,[40112]: 0,[40113]: 0,[40114]: 0,[40115]: 0,[40116]: 0,[40117]: 0,[40118]: 0,[40119]: 0,[40120]: 0,[40121]: 0

error : /
```

Error in 123solar.err :
`PHP Warning:  Division by zero in /var/www/123solar/scripts/protocols/abbuno.php on line 84
`

jeanmarc77, 2022-03-10:
oh.. now it's should be ok 
[test.txt](https://github.com/jeanmarc77/123solar/files/8225661/test.txt)
?

flanesi, 2022-03-11:
No, don't work.

Error in 123solar.err :
`PHP Warning:  trim() expects parameter 1 to be string, array given in /var/www/123solar/scripts/123solar.php on line 506`

jeanmarc77, 2022-03-11:
Hi,
Hard to test without having access, now it should be ok.. i guess ;)
[test.txt](https://github.com/jeanmarc77/123solar/files/8233462/test.txt)

flanesi, 2022-03-11:
I'll try it tomorrow, inverter off.

But there still seem to be errors:
```
modpoll -m tcp -r 41123 -c 26 -0 -a 1 -i -1 192.168.2.22 return : Error on Modpoll: 1st command modpoll 3.10 - FieldTalk(tm) Modbus(R) Master Simulator,Copyright (c) 2002-2021 proconX Pty Ltd,Visit https://www.modbusdriver.com for Modbus libraries and tools.,,Protocol configuration: MODBUS/TCP, FC3,Slave configuration...: address = 1, start reference = 41123 (PDU), count = 26,Communication.........: 192.168.2.22, port 502, t/o 1.00 s, poll rate 1000 ms,Data type.............: 16-bit register, output (holding) register table,Word swapping.........: Slave configured as big-endian word machine,

error : /
```

Unfortunately I can't give you access. I currently have a mobile connection under NAT.

flanesi, 2022-03-15:
don't work.

Test from admin interface:
```
modpoll -m tcp -t4:int -r 40094 -c 1 -0 -a 1 -i -1 192.168.2.22 return : Array

error : /
```

Error in 123solar.err :
`PHP Warning:  trim() expects parameter 1 to be string, array given in /var/www/123solar/scripts/123solar.php on line 506`

flanesi, 2022-03-15:
true:
```
PHP Notice:  Undefined variable: invt_num in /var/www/MyScripts/atest.php on line 50
PHP Notice:  Undefined variable: PHASE in /var/www/MyScripts/atest.php on line 50
PHP Notice:  Array to string conversion in /var/www/MyScripts/atest.php on line 114
It's NOK Array
```

false:
```
PHP Notice:  Undefined variable: invt_num in /var/www/MyScripts/atest.php on line 50
PHP Notice:  Undefined variable: PHASE in /var/www/MyScripts/atest.php on line 50
PHP Notice:  Array to string conversion in /var/www/MyScripts/atest.php on line 114
It's NOK Array
```

jeanmarc77, 2022-03-15:
And that one ?
[atest.txt](https://github.com/jeanmarc77/123solar/files/8255677/atest.txt)

jeanmarc77, 2022-03-15:
ok, so i think it's ok with that one
[abbuno.txt](https://github.com/jeanmarc77/123solar/files/8256061/abbuno.txt)

flanesi, 2022-03-18:
ok, now everything seems to work.

Only one problem remains. But it does not involve problems.
123solar starts before the inverter.
modpoll generates the following error in the log:
`Can't reach server/slave! Check TCP/IP and firewall settings.`
there are 4-5 lines of this error every time the inverter is switched on in the morning

jeanmarc77, 2022-03-18:
Great. For the errors, it's in data/invt1/infos/events.txt ? modpoll.err ? /data/invt1/errors/ ? or even /data/123solar.err ?

flanesi, 2022-03-18:
Error in /data/123solar.err :
`Can't reach server/slave! Check TCP/IP and firewall settings.`

Error in /data/invt1/errors/*.txt
`Error on Modpoll: 1st command modpoll 3.10 - FieldTalk(tm) Modbus(R) Master Simulator,Copyright (c) 2002-2021 proconX Pty Ltd,Visit https://www.modbusdriver.com for Modbus libraries and tools.,,Protocol configuration: MODBUS/TCP, FC3,Slave configuration...: address = 1, start reference = 41123 (PDU), count = 26,Communication.........: 192.168.2.22, port 502, t/o 1.00 s, poll rate 1000 ms,Data type.............: 16-bit register, output (holding) register table,Word swapping.........: Slave configured as big-endian word machine,`

Ather problem but not script dependent.

Power is read with a scale factor of 10 (modbus setting which I have no idea how to change - register 40085 = 1)
Therefore we have a rounding to 10w, which makes the graph very "oscillating"
[image]

With this other script the power is read very accurately, but it is much slower (10 times slower, ~0.6 sec vs 0.06 sec) than the modpoll reading. 
The reading is extracted by accessing the web interface of the inverter.
[GetABBData.txt](https://github.com/jeanmarc77/123solar/files/8306093/GetABBData.txt)

jeanmarc77, 2022-03-18:
123solar.err, not an issue. It's normal as debug mode isn't intended to be permanent.
Errors in /data/invt1/errors/*.txt, that's the aim of that folder. Get all details.

KWHT not precise, that's annoying it's a requirement to have a precise meter (https://github.com/jeanmarc77/123solar/wiki/Protocols)

Maybe you can work on a script that fetch GetABBData.txt just before before the 5 min sample. Like it's :59 to :01 query GetABBData, then query ABBuno. That's should be easy to do.
That say getting sample every seconds is sufficient, especially on ethernet. More can "saturate" your LAN.

closed 2022-03-18 by jeanmarc77