jeanmarc77/123solar #10: bosswerk mi600

dr-ni, 2022-04-23:
I've made a simple bash interface for the Bosswerk Inverter mi600
Can you add a scripts/protocol.php for it?
These values are available:
webdata_now_p
webdata_today_e
webdata_total_e

Examples bash calls
```
pi@rp4:~ $ mi600 solariv admin myPasswd webdata_now_p
421 W
pi@rp4:~ $ mi600 solariv admin myPasswd webdata_today_e
2.83 kWh
pi@rp4:~ $ mi600 solariv admin myPasswd webdata_total_e
237.8 kWh
```

dr-ni, 2022-04-23:
https://github.com/dr-ni/mi600.git

jeanmarc77, 2022-04-24:
Hi,
You can check [how to do so ](https://github.com/jeanmarc77/123solar/wiki/Protocols).
If you have problem can provide raw data returned by your app ? First query should return "yourprotocol.php" values, second _startup.php values, and third _check.php values.

<username> <password> shall be passed via "Communication options"

dr-ni, 2022-04-24:
The script mi600 can only supply the above mentioned three values:
webdata_now_p
webdata_today_e
webdata_total_e
It's returning all results as string.
Other neccessary values shold possibly delivered as a fixed dummy.

Can you add an example protocol that is only feeded with fixed dummy values?
I've tried to modify solarlog.php but was not working only the test in the Inverter section wored.

dr-ni, 2022-04-24:
[image]
[image]
[image]

dr-ni, 2022-04-24:
Have now started my own dummy example:
[example.php.txt](https://github.com/jeanmarc77/123solar/files/8549862/example.php.txt)
[example_checks.php.txt](https://github.com/jeanmarc77/123solar/files/8549863/example_checks.php.txt)
[example_startup.php.txt](https://github.com/jeanmarc77/123solar/files/8549864/example_startup.php.txt)

is this correct?

jeanmarc77, 2022-04-24:
I need all data value return by your app with explanation

dr-ni, 2022-04-24:
my app can get these solar grid-sided values (one phase):
- webdata_now_p: actual solar power (feeding to grid)
command:
`/usr/local/bin/mi600 solariv admin myPasswd webdata_now_p`
answer string:
`421 W`
- webdata_today_e: cumulative daily solar energy (daily feeded until now to grid)
command:
`/usr/local/bin/mi600 solariv admin myPasswd webdata_today_e`
answer string:
`2.83 kWh`
- webdata_total_e: cumulative total solar energy (feeded until now to grid)
command:
`/usr/local/bin/mi600 solariv admin myPasswd webdata_total_e`
answer string:
`187 kWh`

jeanmarc77, 2022-04-24:
Your total kWh counter isn't precise enough => KWHT is the total counter (a Wh precision is needed)
So it is useless to write a protocol for that one, 123s ain't suitable unless you can have more precision

dr-ni, 2022-04-24:
We do'nt need these counters. We can count the webdata_now_p

jeanmarc77, 2022-04-24:
even though you'll miss some data but that's not that important.
You need to have 3 digits behind the comma (2.83x) can you do so ?

dr-ni, 2022-04-24:
I could supply wh precision why is this precision needed?

jeanmarc77, 2022-04-24:
Graph avg power is calculated from that

dr-ni, 2022-04-24:
what is missing in my dummy example?

jeanmarc77, 2022-04-24:
all cmd request ;)
Please read carefully the wiki all is explained

dr-ni, 2022-04-24:
can I also use dummy values for that?
where do I have to add the cmd dummys?

jeanmarc77, 2022-04-24:
[example.php.txt](https://github.com/jeanmarc77/123solar/files/8550148/example.php.txt)

something like that

dr-ni, 2022-04-25:
Ok seems to be working. I'll test it a few days and will then integrate the new counter and all the mi600 bash stuff and curl call into a mi600.php so that your 123Solar logger can fetch all data from the Bosswerk Inverter without any additional app

jeanmarc77, 2022-04-26:
Great, keep me inform

dr-ni, 2022-04-28:
Have now started to hack in my mi600 protocol:
https://github.com/dr-ni/123solar_mi600/blob/main/scripts/protocols/mi600-webif.php
How should I write to DEBUG and how can I throw a not connected alarm which is visible in the UI?

jeanmarc77, 2022-04-29:
looks good.
Normal debug is 2> (stderr), that won't throw things in the UI

jeanmarc77, 2022-04-29:
You can also put 

if ($DEBUG) {
file_put_contents("/var/www/html/dt.txt", $SDTE.": Pnow=".$Pnow." P=".$P." dt=".$DT." E=".$Etotal." Err=".$ERR."\r\n", FILE_APPEND);
}

dr-ni, 2022-05-02:
see also:
https://github.com/jeanmarc77/123solar/pull/11

jeanmarc77, 2022-05-02:
Thanks for the pull

closed 2022-05-02 by jeanmarc77