jeanmarc77/123solar #13: Energy summation not correct?

dr-ni, 2022-05-04:
I have some problems with the energy count. It is counting too much. Is your curve smoothing algorithm working correct?

[image]

[slog.txt](https://github.com/jeanmarc77/123solar/files/8620106/slog.txt)

jeanmarc77, 2022-05-04:
Sure, the avg power is calculated between 2 samples with a second precision

dr-ni, 2022-05-04:
I've found the main problem:
https://github.com/jeanmarc77/123solar/pull/14

but still I get different curves with two 123solar running paralel

dr-ni, 2022-05-04:
Another question:
Why is the power-meter on the main page often going up to Pnow and down to 0 - even if continuous power is feeded?

jeanmarc77, 2022-05-04:
It mean your app return 0 or your last live sample is 30sec too old.
Please don't use issue for help/question but only for issue/improvement

closed 2022-05-04 by jeanmarc77