jeanmarc77/123solar #17: logger stopped with this error

M0stie, 2022-05-16:
PHP Fatal error: Uncaught ValueError: fread(): Argument #2 ($length) must be greater than 0 in /srv/http/123solar/scripts/123solar.php:650
Stack trace:
#0 /srv/http/123solar/scripts/123solar.php(650): fread()
#1 {main}
thrown in /srv/http/123solar/scripts/123solar.php on line 650

Its been running since Janaury with no issues and nothing has been updated, it just stopped on Saturday

jeanmarc77, 2022-05-16:
Hi,
Never had that issue, but it is fixed in main branch
https://github.com/jeanmarc77/123solar/commit/94e54d649fe6491eb1c2c38f3e05f61e6501eb5c

closed 2022-05-16 by jeanmarc77

M0stie, 2022-05-19:
Thank you, it’s been working now for a couple of days 

Graham

jeanmarc77, 2022-05-19:
I understand, it's because you use a GIT clone, instead of using the stable release