jeanmarc77/123solar #18: PHP errors

halfagascan, 2022-05-21:
123Solar 1.8.4.4
PHP Notice:  Trying to access array offset on value of type null in /srv/123solar/config/pvoutput/extended/extended1.php on line 4
PHP Notice:  Undefined variable: array2 in /srv/123solar/config/pvoutput/extended/extended1.php on line 5
PHP Notice:  Trying to access array offset on value of type null in /srv/123solar/config/pvoutput/extended/extended1.php on line 5
Web page is stuck on 13/05/2022, will not refresh.
I was going to roll back to a previous version, but I have no previous version and am unable to locate one.
Thanks

halfagascan, 2022-05-21:
using SBFspot V3.4.0

jeanmarc77, 2022-05-21:
Hi, 
Extended data send by defaut by 123s are grid phase voltage. I suppose you have not a triphased system.
I should put null for all extended values by default.

Either edit extended1.php for your need or disable the tick in "PVoutput configuration".
BTW, if you update via 123s, you always have a backup in _INSTALL dir and nothing on that have been change since previous version

closed 2022-05-21 by jeanmarc77