jeanmarc77/123solar #21: Error by pressing "On/Off" Button

DirkAlbi, 2022-07-19:
Since a few weeks, I get an Error "Fehler Verbindung unterbrochen",wenn I press the Button "On/Off"-Button.
I log in as admin and get to admin.php, the On button is activated. If I try to disable it I get an error page that the connection was lost. When I press the "Back" button, the page flickers briefly and I end up back in "Admin.php". What information is required for targeted troubleshooting?
RaspberryPi 3, Debian Bulleseye, Apache 2.4, php 7.4

Thanks a lot for help.

jeanmarc77, 2022-07-19:
Hi,
Do you use lastest 123solar version ?
Check if 123s run : ps -ef | grep 123solar
Any hints into Help and debugger ?
Make sure your HTTP user can write into /dev/shm/

DirkAlbi, 2022-07-19:
Hello, Jeanmarc,
thanks you for answer. I use Version 1.8.4.4. 
ps -ef | grep 123solar reports "pi       23784 23770  0 22:08 pts/0    00:00:00 grep --color=auto 123solar". 
Debug is acivated. Where can I see Errors?

jeanmarc77, 2022-07-20:
It seem like there a problem with /dev/shm/.
You can get details via admin > /Help and debugger

DirkAlbi, 2022-07-21:
Thank you Jeanmarc,
now it is working correctly. :-)
/dev/shm was okay. In "Help and Debugger" I found the notice "Since PHP 7.4 there is hardening options. Current version is 7.4.30. Allow to use your com. devices by setting PrivateDevices=false in php-fpm.service. (e.g. 'systemctl edit --full php-fpm.service')  and 'systemctl restart php7.4-fpm". After I take this, the 123solar war running fine.
I´m happy and must donate for you and your for good software.
Best wishes, Dirk Albers

jeanmarc77, 2022-07-21:
Great, cheers Dirk

closed 2022-07-21 by jeanmarc77