jeanmarc77/123solar #22: PHP8.0 Compatibility

escix, 2022-07-31:
Hi,
Is this code compatible with PHP8.0.
I am getting lots of error with PHP8.0 where is was working fine with PHP7.2

jeanmarc77, 2022-07-31:
Sure, i run with PHP 8.1.8.
What are the error ? Check hints into '[Help and debugger]'

escix, 2022-07-31:
the script doesn't create the 123solar.pid file.
Also I had the following errors:
PHP Warning:  Trying to access array offset on value of type null in...

Checking running softwares is also blank.

jeanmarc77, 2022-07-31:
What about admin/fperms.php ?

escix, 2022-07-31:
All OK

jeanmarc77, 2022-07-31:
duh.. maybe enable debug, restart 123s, check also your webserver error log and send me more details

escix, 2022-07-31:
I run 123Solar in FreeBSD, which I run before too but older version in Raspberry PI.
Now moved to x64 thinclient and upgraded to latest version with PHP8.0.
The reading works still, but the button in the admin panel stays off.

jeanmarc77, 2022-07-31:
do you have a ../scripts/123solar.pid file ? I guess not, that's the issue

escix, 2022-07-31:
Now fixed the ps command in admin/help.php file for FreeBSD.
Now the Running Softwares in the help.php shows the running processes.

The 123solar.pid file is not getting created when starting. not sure why?

jeanmarc77, 2022-07-31:
maybe the command is unappropriate for FreeBSD, in admin line 118 : ps -ef | grep $PID | grep 123solar.php

escix, 2022-07-31:
perfect, working now.
Cheers

jeanmarc77, 2022-07-31:
what was the command ? 

i guess scripts/distros/ commands are not good for FreeBSD. Mind checking them and make a new one ?
I could set the PID command return different by setting distro in admin panel.

escix, 2022-07-31:
ps -auxw to display all processes

jeanmarc77, 2022-07-31:
ok and what for 
$UPTIME=
$CPUUSE=
$MEMTOT=
$MEMUSE=
$MEMFREE=
$DISKUSE=
$DISKFREE

escix, 2022-07-31:
Except $MEM* everything else works as expected.
For memory info a pkg needs to beinstalled. I think it is memfree or freemem or similar. Sorry I forgot - which I don't have installed.

jeanmarc77, 2022-07-31:
ok, so tell me if you know. I work by implementing PID via distro
thanks

closed 2022-07-31 by jeanmarc77

escix, 2022-07-31:
Could you use something like this:

sysctl -a | grep 'real memory'
sysctl -a | grep 'avail memory'

jeanmarc77, 2022-07-31:
i've push on git, it will work on next stable update.
Mind checking https://github.com/jeanmarc77/123solar/blob/main/scripts/distros/FreeBSD.php ?

escix, 2022-07-31:
$CPUUSE="ps aux|awk 'NR > 0 { s +=$3 }; END {print \"cpu %\",s}' | awk '{ print $3 }'";

this doesn't work

error:
awk: syntax error at source line 1
 context is
        NR > 0 { s +=$3 }; END {print >>>  \ <<< "cpu %\",s}
awk: illegal statement at source line 1

jeanmarc77, 2022-07-31:
please provide the full working FreeBSD.php, i'll push it on git

escix, 2022-07-31:
$CPUUSE="ps aux | awk 'NR > 2 { s +=$3 }; END {print "cpu %",s}' | awk '{print $3 }'"

escix, 2022-07-31:
How does the distro files work?
Does it automatically get the info or does it need to be modified in each php files?

for eg. in info.php there is this command which won't work in FreeBSD:
cat /proc/cpuinfo | grep 'Processor' | head -n 1

same in admin.php as stated before.

jeanmarc77, 2022-07-31:
Not all commands are define there like cpuinfo, i can add them via distros files.
It simply PHP exec the commands, just tell me what's work on BSD.

I've modify admin.php https://github.com/jeanmarc77/123solar/blob/main/admin/admin.php

escix, 2022-07-31:
$CPUUSE="ps aux | awk 'NR > 0 { s +=$3 }; END {print "\cpu %",s}' | awk '{ print $3 }'";

When I use the above command i get the following error:
FastCGI sent in stderr: "PHP message: PHP Parse error:  syntax error, unexpected fully qualified name "\cpu" in

But it works on command line if I execute as user www.

% ps aux | awk 'NR > 0 { s +=$3 }; END {print "cpu %",s}' | awk '{ print $3 }'
0
%

jeanmarc77, 2022-07-31:
You miss a \, that's
$CPUUSE="ps aux | awk 'NR > 0 { s +=$3 }; END {print "\cpu %\",s}' | awk '{ print $3 }'";

escix, 2022-07-31:
I am getting an error when I put that backslash.

What should be the output from the above command?
Is that just a number or cpu % 10 ?

jeanmarc77, 2022-07-31:
It should return a value like 25 which mean 25%

don't forget to remove the escape backslash if in CLI
`ps aux|awk 'NR > 0 { s +=$3 }; END {print "cpu %",s}' | awk '{ print $3 }'`

escix, 2022-07-31:
the one in the Ubuntu works.

Does it only shows the cpuuse for 123solar or for all processes?

escix, 2022-07-31:
> 

That's it I was checking in the CLI and not working, but it was working in the web

jeanmarc77, 2022-07-31:
It's total CPU usage like you can see via top.

CPU/men usage of 123s is seen on help page, lines 47 & 48
`	$cpu = exec("ps -p $PID -o %cpu | tail -1 | awk '{print $1}'");
	$mem = exec("ps -p $PID -o %mem | tail -1 | awk '{print $1}'");
`

escix, 2022-07-31:
yes you are correct. It is the cpuuse by www user.

jeanmarc77, 2022-07-31:
Maybe this 
`
$loads = sys_getloadavg();
$core_nums = trim(shell_exec("grep -P '^processor' /proc/cpuinfo|wc -l"));
$CPUUSE = round($loads[0]/($core_nums + 1)*100, 2);

`