jeanmarc77/123solar #24: Problems with more than one Bosswerk MI-600 inverter in the same grid

soe135, 2022-08-03:
Hello all,
I'm using more than one MI-600 inverter in the same grid with 123solar. As I have found out so far is that the recent code in /scripts/protocols/mi600-webif.php is only working correctly with one inverter. In addition, there were often problems in reading the values from the webif of the inverter. So I had many 'jumping' power-values (spikes) in the 'index-chart' of the 123Solar logger. That's why I have adapted and rewritten some parts of the file mi600-webif.php. Now the code is working for me without any problems since a week and I decided to publish the changes for other users. If you find my changes helpful please feel free to implement these into your repository. I have attached my modified file under the following link

[mi600-webif.zip](https://github.com/jeanmarc77/123solar/files/9249541/mi600-webif.zip)

Best regards, soe135

jeanmarc77, 2022-08-03:
Hi Soe,

Thanks for you work, i believe you should push your code into https://github.com/dr-ni/123solar_mi600.
Then it will surely push the code into 123s

It is wiser to use this as server path 
$SRVDIR = $_SERVER['DOCUMENT_ROOT']; 
$LOGFILE = "$SRVDIR/123solar/data/invt".$invt_num."/mi600.log";
..
Because some have /srv/http, /var/www/html or whatever

Thanks

soe135, 2022-08-03:
Hi jeanmarc77,

Thanks for your reply and the suggestion. I will change the file variables as you mentioned and then push the code into https://github.com/dr-ni/123solar_mi600.

Best regards, soe135

closed 2022-08-03 by jeanmarc77