jeanmarc77/123solar #27: changing meter size when current is higher than 10A

dr-ni, 2022-08-07:
if the resolution is <= 720p the meter size is always changin because of an unwanted linebreak

possible fix: change VDC to V to save space
[image]
[image]

jeanmarc77, 2022-08-07:
Hi,
I believe it's wiser to keep VDC, as people can mess up with array and grid voltage.
It can change it to ⎓ light be cooler.
Nobody use <720p ;)

dr-ni, 2022-08-07:
ok keep me informed so I can test it

jeanmarc77, 2022-08-07:
ok you can give it a try, even though 123s interface is suppose to be see on a pc monitor.
There is m.php, which is supposed to be mobile or minimale interface. (i haven't work on it much)

dr-ni, 2022-08-08:
perfect :-)

closed 2022-08-08 by dr-ni