jeanmarc77/123solar #32: Time out: can't retreive https://123solar.org/latest_version.php

halfagascan, 2022-08-20:
from 123Solar Administration, select 123solar 1.8.4.3, update, results in the above error. I see 1.8.4.4 as the latest, on github.
In addition, last week, a transformer exploded in the area, lost power, after sometime power was restored, all internet was good, reestablished interconnection, via openbsd, wg-quick, no issues, there were no issues with connection to internet, via linux, win10, or android devices, all was normal, with the exception of 123solar. From a linux box to web of 123solar/, produced absolutely zero results, the main web site would load and display, no graphs, no historical data, other web pages displayed relevant info, however extremely slow to load,on the order of 10-15 minutes to load a single page, a refresh or reload required 10-15 minutes to load.
From an android device to the webservice of 123solar, I experienced zero issues, no lag on reload, or refresh, it showed current and relevant information. I experienced this issue using firefox, dolphin, konqueror on linux, and win10. I booted from a usb device, using 
Q4os, same issue, no data displayed
systemctl restart php7.4-fpm.service nginx did not solve the issue.
No errors in nginx logs, or access logs.
What DID resolve the error, was to reconnect to the VPN provider, 123solar loaded very quickly, displays and updates the website.
So what outside connections does 123solar REQUIRE?
Thanks

halfagascan, 2022-08-20:
forgot to add, the invertor is reporting a temp of Invert. Temp. -21,474,836.0 °c.
I had reported this in the past and you provided a solution, sadly the website is gone and am unable to solve this issue

jeanmarc77, 2022-08-21:
Hi,
You should have update on time, i warn months ago i would close the website and move to git.
You have to manualy update, check the wiki.

One problem at once please.
I don't know what happen to your logger, it's not 123s related. It only need port 80 to be fwd

closed 2022-08-21 by jeanmarc77