jeanmarc77/123solar #35: PHP Fatal error:  Uncaught ValueError

kitenski, 2022-10-31:
I noticed my long running 123solar wasn't up, everytime I restarted it, it died quickly.

Turning on debug shows the following:

#* 31/10/2022 12:03:08	Starting 123Solar debug (1461)

PHP Fatal error:  Uncaught ValueError: fread(): Argument #2 ($length) must be greater than 0 in /srv/http/123solar/scripts/123solar.php:650
Stack trace:
#0 /srv/http/123solar/scripts/123solar.php(650): fread()
#1 {main}
  thrown in /srv/http/123solar/scripts/123solar.php on line 650

jeanmarc77, 2022-10-31:
Hi,
You probably have a empty file in data/invtX/msgqueue/, try to remove it

kitenski, 2022-10-31:
thanks, indeed, that has fixed it!

There were many empty files!

1666767540R.txt  1667215800R.txt  1667217180R.txt  1667218200R.txt

1666872840R.txt  1667216220R.txt  1667217720R.txt  1667218380R.txt

1667213940R.txt  1667216340R.txt  1667217780R.txt

closed 2022-10-31 by jeanmarc77

kitenski, 2022-10-31:
UK autumn at it's finest, or old panels maybe?

[image]

jeanmarc77, 2022-10-31:
A low Riso can happen because of a poor ground (too dry) and, ileak when it's stormy weather, i think it's because of static electricity.
Anyway, your message files shouldn't have to be empty, maybe your logger was writing when you shut it down or a SDcard that is slowly dying