jeanmarc77/123solar #37: Wrong read on three Phases

Alexandros89, 2022-11-24:
Hi, i have two aurora pvi10 and when the energy is less than 1 kwh the read is very wrong. This problem also make wrong statistics with the consume meter

jeanmarc77, 2022-11-24:
Hi,
I am not the author of 'aurora' com app. You need to have a wh precision.
What aurora -b -a2 -c -T -d0 -e -3 /dev/ttyXXX return ?

Alexandros89, 2022-11-27:
pi@raspberrypi:~ $ aurora -b -a2 -c T -d0 -e -3 /dev/ttyUSB0

aurora -b -a2 -c T -d0 -e -3 /dev/ttyUSB0

szttyDevice: T
yDelay:     1
yTimeout    0 mS
yMaxRunTime ~
devLCKfile: <(null)>
devLCKfileNew: <(null)>
Got Params

RunTime     20221127-07:36:46 v1.9.4
Endian    : Little
tm_gmtoff : yes
PID       : 13051

Attempting to get lock on Serial Port T...

20221127-07:36:46: aurora: Problem locking serial device, can't open lock file: (null) for write.

pi@raspberrypi:~ $ sudo aurora -b -a2 -c T -d0 -e -3 /dev/ttyUSB0

aurora -b -a2 -c T -d0 -e -3 /dev/ttyUSB0

szttyDevice: T
yDelay:     1
yTimeout    0 mS
yMaxRunTime ~
devLCKfile: <(null)>
devLCKfileNew: <(null)>
Got Params

RunTime     20221127-07:37:06 v1.9.4
Endian    : Little
tm_gmtoff : yes
PID       : 14165

Attempting to get lock on Serial Port T...

20221127-07:37:06: aurora: Problem locking serial device, can't open lock file: (null) for write.

pi@raspberrypi:~ $ aurora -b -a2 -c -T -d0 -e -3 /dev/ttyUSB0

aurora -b -a2 -c -T -d0 -e -3 /dev/ttyUSB0

szttyDevice: /dev/ttyUSB0
yDelay:     1
yTimeout    0 mS
yMaxRunTime ~
devLCKfile: </var/lock/LCK..ttyUSB0>
devLCKfileNew: </var/lock/LCK..ttyUSB0.16281>
Got Params

RunTime     20221127-07:37:43 v1.9.4
Endian    : Little
tm_gmtoff : yes
PID       : 16281

Attempting to get lock on Serial Port /dev/ttyUSB0...
Checking for lock
PID: 16281 COMMAND: "aurora" LckPID: 16281 LckCOMMAND: "aurora" LckPIDcommand "aurora" = me
Appears we got the lock.

Opening Serial Port /dev/ttyUSB0...  Serial Port /dev/ttyUSB0 successfully opened.
Configuring serial device... Flushing unread data first...  Success!
Flushing serial device buffer... Success!

Comm Check: Let's see if the Aurora is listening...
Elapsed time since last comm 0 us
[... 683 lines cut ...]
Complete 20221127-07:37:43

pi@raspberrypi:~ $
pi@raspberrypi:~ $ pi@raspberrypi:~ $ aurora -b -a2 -c -T -d0 -e -3 /dev/ttyUSB0
yDelay:     1
yTimeout    0 mS
yMaxRunTime ~
devLCKfile: </var/lock/LCK..ttyUSB0>
devLCKfileNew: </var/lock/LCK..ttyUSB0.16281>
Got Params

RunTime     20221127-07:37:43 v1.9.4
Endian    : Little
tm_gmtoff : yes
PID       : 16281

Attempting to get lock on Serial Port /dev/ttyUSB0...
Checking for lock
PID: 16281 COMMAND: "aurora" LckPID: 16281 LckCOMMAND: "aurora" LckPIDcommand "aurora" = me
Appears we got the lock.

Opening Serial Port /dev/ttyUSB0...-bash: pi@raspberrypi:~: comando non trovato
  Serial Port /dev/ttyUSB0 successfully opened.
Configuring serial device... Flushing unread data first...  Success!
Flushing serial device buffer... Success!

jeanmarc77, 2022-11-27:
97769.392 is your total kWh counter, it is precise enough. You mean: if it's under 1kWh a day, it dosen't count properly ?

Alexandros89, 2022-11-27:
My problem is the instant production when the production is less then 1 kwh

jeanmarc77, 2022-11-27:
Check in help and debugger the 123s_LIVEMEMORY.json values when it happen

Alexandros89, 2022-11-27:
Ok, later when it will happen i'll check it

Alexandros89, 2022-11-27:
[image]
 This is the problem

jeanmarc77, 2022-11-27:
don't know what Serra is.
which version do you use ?
Please report 123s_LIVEMEMORY.json

Alexandros89, 2022-11-27:
[image]
This?
I use the 1.8.4.3

jeanmarc77, 2022-11-27:
yeah but when it happen, here your logger are asleep

Alexandros89, 2022-11-29:
#* 29/11/2022 15:16:09	Stopping 123Solar debug (16831)

#* 29/11/2022 15:16:12	Starting 123Solar debug (27993)

{"info":"no active devices to send to","status":1,"request":"346fe3ea-f598-4b42-9227-0c81a36a2778"}{"token":"invalid","errors":["application token is invalid"],"status":0,"request":"67b40bdd-a8b6-4ac1-b541-09ee3ef27d1a"}{"info":"no active devices to send to","status":1,"request":"e26f6190-52b1-49e1-bf71-54a37a0adedf"}{"token":"invalid","errors":["application token is invalid"],"status":0,"request":"5db82480-3996-4ae2-a5fc-778bb8f46cf4"}{"token":"invalid","errors":["application token is invalid"],"status":0,"request":"ad24337c-4374-4e2c-9995-43c23b07e3c4"}{"info":"no active devices to send to","status":1,"request":"9deafc04-99bc-4a08-98d7-6d37032eec29"}{"token":"invalid","errors":["application token is invalid"],"status":0,"request":"ec614a7f-7776-4c6e-bfac-b71eca6de82a"}

Alexandros89, 2022-11-29:
yes i set it

jeanmarc77, 2022-11-29:
Which protocol are you using ? And, please report 123s_LIVEMEMORY.json

Alexandros89, 2022-11-29:
I'm using the aurora protocol with options -Y6 -I5
do you want a copy of 123s_LIVEMEMORY.json?

closed 2022-11-29 by jeanmarc77

Alexandros89, 2022-11-29:
Why have you close?

jeanmarc77, 2022-11-29:
can't debug without details

Alexandros89, 2022-11-29:
what details you need?

jeanmarc77, 2022-11-29:
GxPx values when that happen

Alexandros89, 2022-11-29:
where i can find?

Alexandros89, 2023-04-01:
i have found the problem, the wrong read is because 123solar use the grid values that is not real in the early morning, the correct read is the arrays values
[image]

jeanmarc77, 2023-04-01:
If the values returned are not correct i can't do much about it.
You can contact Curt from aurora if he know how to fix that but i am aware some inverters report wrong values.