jeanmarc77/123solar #42: no help found for using multiple inverters

dr-ni, 2023-02-17:
Can you add some info for protocols setup for multiple inverters

dr-ni, 2023-02-17:
It's not clear for me how to supply `{'ADR'.$invt_num}` variable and `${'COMOPTION'.$invt_num}` when using two inverters - for example two mi600 with mi600-webif protocol.
- One mi600: "mi600-no1-ip username password" can be supplied in the option field
Where or how can the second  inverter "mi600-no2-ip username password"  be supplied?

The help doesn't tell me anything about such setup.

jeanmarc77, 2023-02-17:
Your $HOST variable should rather be ${'ADR'.$invt_num} and not passed as a com option

dr-ni, 2023-02-17:
so how exactly in my case?

mi600-no1-ip username1 password1
mi600-no2-ip username2 password2

jeanmarc77, 2023-02-17:
Pass username and password as ${'COMOPTION'.$invt_num};

```
$OPTIONS = ${'COMOPTION'.$invt_num};
list ($USER, $PASSWD) = explode(' ', $OPTIONS, 2);
$URL = 'http://'.${'ADR'.$invt_num}.'/status.html';
```

dr-ni, 2023-02-17:
can yo give a hard coded example for th setup above, it's still unclear for me

jeanmarc77, 2023-02-17:
```
<?php
# Stand-alone protocol for Bosswerk MI600 (Deye SUN600G3-EU-230) inverter
# It's now also working with more than one inverter in a grid.
# Author: https://github.com/dr-ni/123solar_mi600, adapted by SOE135
# License GPL-v3+
#
# Please supply '<host> <user> <passwd>' for your MI600 in Admin -> Inverter(s) configuration
# Use the field 'Communication options'

if (!defined('checkaccess')) {die('Direct access not permitted');}
$CMD_RETURN = ''; // always initialize
$MATCHES = '';
$ERR = "0";
$SDTE = date("Ymd H:i:s");
$OPTIONS = ${'COMOPTION'.$invt_num};
list ($USER, $PASSWD) = explode(' ', $OPTIONS, 2);
$URL = 'http://'.${'ADR'.$invt_num}.'/status.html';
// ##### The location of some files has changed in this patch! ####
// ##### Please move your old lastKWHtotal.dat file manually from inverters data-directory to the errors subdirectory! 
$LOGFILE = "$INVTDIR/errors/mi600.log";
$MI600_DATAFILE = "$INVTDIR/errors/mi600.dat";
$LAST_KWHTOTAL_FILE = "$INVTDIR/errors/lastKWHtotal.dat";
// strings
$I1V = null;
$I1A = null;
$I1P = null;
$I2V = null;
$I2A = null;
$I2P = null;
$I3V = null;
$I3A = null;
$I3P = null;
$I4V = null;
$I4A = null;
$I4P = null;
// inverters
$FRQ = null;
$EFF = null;
$INVT = null;
$BOOT = null;
$KWHT = null;
// grids
$G1V = null;
$G1A = null;
$G1P = null;
$G2V = null;
$G2A = null;
$G2P = null;
$G3V = null;
$G3A = null;
$G3P = null;
// variable names to store process values for the specific inverter
$LastP = 'MI600INVT'.$invt_num.'_LASTP';
$LastPTS = 'MI600INVT'.$invt_num.'_LASTPTS';
$LastKWHT = 'MI600INVT'.$invt_num.'_LASTKWHT';
// other needed variables
$MaxRetryCount = 5;
$RetryCounter = 0;
$DataOK = 0;
$P = (float) 0;
$Dt = 0;
$MinSecondsBetweenMeasurements = 10;
$Now = time();
$KWHTDifference = (float) 0; // difference between the calculated KWHT-value and the (real) KWHT-value from inverters webif
$KWHTCorrectionFactor = (float) 0; // standard behaviour, no correction needed

// initializing process variables
if (!isset($P)) $P = 0;
if (!isset($$LastP)) $$LastP = (float) 0; // last determined power value for this inverter
[... 112 lines cut ...]
  if ($KWHTCorrectionFactor > 0) {
    file_put_contents("$LOGFILE", $SDTE.": Ongoing KWHT-correction, KWHTDifference=".$KWHTDifference." KWHTCorrectionFactor=".$KWHTCorrectionFactor."\r\n", FILE_APPEND);
  }
  file_put_contents("$LOGFILE", $SDTE.": P=".$P." P_LAST=".$$LastP." KWHTotal=".$$LastKWHT." NewKWHTDelta=".$NewKWHTDelta." Err=".$ERR."\r\n", FILE_APPEND);
}  

$INVT= 0; // temperature inverter fixed dummy
$BOOT = 0; // temperature dc/dc booster fixed dummy
$I1V = (float) 36; // udc fixed dummy
$I1P = (float) $P;
$I1A = (float) round($I1P / $I1V, 2);
$G1P= (float) $P; // P-AC
$G1V= (float) 230; // U-AC fixed dummy
$G1A = (float) round($G1P / $G1V, 2);
$FRQ= (float) 50; // freq  fixed dummy
$KWHT = $$LastKWHT; 

if ($ERR == "0") {
  $RET = 'OK';
} else {
  $RET = 'NOK';
}

?>
```

dr-ni, 2023-02-17:
is this the original protocol?

where are the settings?

cannot find

mi600-no1-ip username1 password1
mi600-no2-ip username2 password2

jeanmarc77, 2023-02-17:
Yeah that's your modified code

config/config_invt1.php will set as $ADR1 no1-ip and $COMOPTION1 username1 password1
config/config_invt2.php will set as $ADR2 no2-ip and $COMOPTION2 username2 password2

dr-ni, 2023-02-17:
so do I have to add these files like this?

`config/config_invt1.php`
```
$ADR1="no1-ip";
$COMOPTION1="username1 password1";

...

```

jeanmarc77, 2023-02-17:
of course not, juste use the interface it will create them

dr-ni, 2023-02-17:
which interface?

jeanmarc77, 2023-02-17:
gosh did you ever use 123s ? http://yourlogger/123solar/admin/

dr-ni, 2023-02-17:
of course but is there something in the wiki about?

its still not clear how to setup multiple inverters

jeanmarc77, 2023-02-17:
can be more simple than that 
123solar/admin/admin_main.php Number of inverter(s) 2
then
123solar/admin/admin_invt.php 
Inverter #1 set IP adress 1 Communication options username1 password1
Inverter #2 set what you want

dr-ni, 2023-02-17:
ahhhhh ok this is what was missing would be nice in the wiki, not everybody knows all details as you

jeanmarc77, 2023-02-17:
Number of inverter(s) is the first thing to look at in Main configuration ;p

closed 2023-02-17 by jeanmarc77