jeanmarc77/123solar #44: Test Communication works, but 123Solar not logging any data

mcpaton, 2023-03-05:
I had 123Solar working nicely on my QNAP NAS. On Feb 22 it seemed to stop working (may have been related to reboot of NAS). The ttyUSB0 has not changed, and when I use the Test Communication button in the Admin area, it returns correct data from the Aurora Inverter (I see this update in realtime when I refresh the page). However when I run the 123Solar service, the web pages are stuck on Feb 22 and do not show any live data.

What do I need to do to get the main data to work?

jeanmarc77, 2023-03-05:
Does 123solar.php runs ? Don't you have a /var/lock/LCK..USB file stuck ?
What returns ls -al /dev/shm/ ?
In debug mode, there is no errror log ?

mcpaton, 2023-03-05:
I'm not sure how to tell if 123solar.php runs? Or how to tell if the usb file is stuck.

bash-4.4# ls -al /dev/shm
total 8
drwxrwxrwt    2 root     root            80 Mar  5 07:50 .
drwxr-xr-x   14 root     root         21480 Mar  5 06:02 ..
-rw-r--r--    1 nginx    www-data       256 Mar  5 07:51 123s_LIVEMEMORY.json
-rw-r--r--    1 nginx    www-data       226 Mar  5 07:51 123s_MEMORY.json

I have restarted in debug mode:
#* 05/03/2023 16:50:01	Starting 123Solar debug (1926)

Memory (/dev/shm/123s_MEMORY.json):
Array ( [awake] => [First1] => 0 [pmotd1] => 0 [pmotdt1] => 1677992549 [pmotdmulti] => 0 [pmotdtmulti] => 0 [status] => 123Solar ZzzZ [invtstat1] => [5minflag1] => [msgq1] => 0 [AWt1] => 1677992549 [AWriso1] => 0 [AWileak1] => 0 [peakotd1] => 0 [peakoat1] => 0 )
Live memory (/dev/shm/123s_LIVEMEMORY.json)
Array ( [SDTE1] => 1677999001 [I1V1] => 0 [I1A1] => 0 [I1P1] => 0 [I2V1] => 0 [I2A1] => 0 [I2P1] => 0 [I3V1] => 0 [I3A1] => 0 [I3P1] => 0 [I4V1] => 0 [I4A1] => 0 [I4P1] => 0 [G1V1] => 0 [G1A1] => 0 [G1P1] => 0 [G2V1] => 0 [G2A1] => 0 [G2P1] => 0 [G3V1] => 0 [G3A1] => 0 [G3P1] => 0 [FRQ1] => 0 [EFF1] => 0 [INVT1] => 0 [BOOT1] => 0 [SSR1] => 0 )

Which shows no data. But the Test Communication shows this data:

aurora -a 2 -c -T -Y3 -d0 -e /dev/ttyUSB0
Arrays values
I1V : 301.955658 V | I1A : 0.646091 A | I1P : 195.090897 W
I2V : 191.440048 V | I2A : 0.650119 A | I2P : 124.458862 W
I3V : V | I3A : A | I3P : W
I4V : V | I4A : A | I4P : W
Grid values
G1V : 241.213562 V | G1A : 1.353939 A | G1P : 311.210297 W
G2V : V | G2A : A | G2P : W
G3V : V | G3A : A | G3P : W
Inverter values
FRQ : 49.986004 Hz | EFF : 97.390251 % | INVT : 41.979527 °
BOOT : 40.722416 ° | KWHT : 61247.475 kWh | Sensor : W/m²

I see on the help/debig page I have a red X next to 
/dev/shm is -NOT- OK  ok to write 
but a green tick next to OK to write.
But this was the case even when it was working OK.

mcpaton, 2023-03-05:
Date and time is correct (Australia/Brisbane)

df -h |grep /dev/shm
shm                      64.0M      8.0K     64.0M   0% /dev/shm

jeanmarc77, 2023-03-05:
is it the same TimeZone in Main configuration ?
123s is sleeping [status] => 123Solar ZzzZ

mcpaton, 2023-03-05:
That’s where I checked the Timezone. Is there somewhere else to check?
The solar system is awake at the moment as it’s daytime.

jeanmarc77, 2023-03-05:
on your system check "timedatectl"
If not correct "timedatectl set-timezone Australia/Brisbane"

mcpaton, 2023-03-05:
Ok the Timezone was wrong. I had to use a different command but now it is showing the correct date and time it is now working. Thanks for your assistance!

closed 2023-03-05 by jeanmarc77