jeanmarc77/123solar #45: Inverter Won't Connect

mcpaton, 2023-04-02:
So looks like my NAS did an update the other night and rebooted. The Serial drivers did not restart, so the USB to RS485 adaptor did not connect back up.

I have restarted it all today, however 123Solar is not showing data from the inverter. If I do a test communications it shows relevant data. 

Is there some init string or something required to get it to start reading data from the inverter? And is there any way to download the data from the inverter that I have lost from the last couple of days? The old Aurora Communicator software used to be able to do this.

Debug mode makes it look like the inverter is asleep (I have checked Timezone etc and it's all correct)

Memory (/dev/shm/123s_MEMORY.json):
Array ( [awake] => [First1] => 0 [pmotd1] => 0 [pmotdt1] => 1680406247 [pmotdmulti] => 0 [pmotdtmulti] => 0 [status] => 123Solar ZzzZ [invtstat1] => [5minflag1] => 1 [msgq1] => 0 [AWt1] => 1680406247 [AWriso1] => 0 [AWileak1] => 0 [peakotd1] => 0 [peakoat1] => 0 )
Live memory (/dev/shm/123s_LIVEMEMORY.json)
Array ( [SDTE1] => 1680407087 [I1V1] => 0 [I1A1] => 0 [I1P1] => 0 [I2V1] => 0 [I2A1] => 0 [I2P1] => 0 [I3V1] => 0 [I3A1] => 0 [I3P1] => 0 [I4V1] => 0 [I4A1] => 0 [I4P1] => 0 [G1V1] => 0 [G1A1] => 0 [G1P1] => 0 [G2V1] => 0 [G2A1] => 0 [G2P1] => 0 [G3V1] => 0 [G3A1] => 0 [G3P1] => 0 [FRQ1] => 0 [EFF1] => 0 [INVT1] => 0 [BOOT1] => 0 [SSR1] => 0 )

Also to note - on the main page. the Sunrise, Sunset and Transit times are wrong (17:35, 23:21, 05:07 respectively)

Thanks

jeanmarc77, 2023-04-02:
It's not an issue with 123s.
Check your **system** time and timezone (date and timedatectl), 123s is sleeping so it don't retrieve data.
You can get previous data from aurora, read the manual

closed 2023-04-02 by jeanmarc77

mcpaton, 2023-04-02:
Both the System Date and Time, the Container Date and Time all read my current time.

jeanmarc77, 2023-04-02:
is your Local time correct with timedatectl ?
did you set correctly PHP TimeZone with 123s admin > main ?

mcpaton, 2023-04-02:
Yes that is set correctly

In the info/debug page it also shows correctly:
You timezone is set to Australia/Brisbane (02/04/2023 17:26:19)

mcpaton, 2023-04-02:
The dashboard instanreneios settings also shows local time correctly, but the data often is incorrect:

Dashboard - 30/03/2023 (18,5 kWh) | 02/04/2023 17:29:56 | 500020001000500200100 ms

jeanmarc77, 2023-04-02:
what timedatectl returns ?

mcpaton, 2023-04-02:
timedatectl doesn't work.
I managed to open another shell into the container and did a "date" command which shows the wrong timezone. I have set this now in the Environment for the container and it's now showing correctly. 
It's now night time so I will need to check again tomorrow.
It's clear the settings in the config page of 123Solar seem to have no effect on the system.

jeanmarc77, 2023-04-02:
Obviously the 123s settings is for PHP, it cannot change your system config