jeanmarc77/123solar #46: Last 20 Days Changed

halfagascan, 2023-04-26:
Something changed with viewing the last 20 days graph, yesterday it was the normal vertical bars, today its a line graph.
[image]

jeanmarc77, 2023-04-26:
Yup see https://github.com/jeanmarc77/meterN/issues/10

halfagascan, 2023-04-26:
Installed new index.php
restart nginx, php-fpm7.4
The dashboard gauge went goofy
Installed php-fpm8.2
start php8.2
guage returns, the last 20 days, no change, still shows a line graph
systemctl stop php-fpm7.4
Results in bad gateway
systemctl start php-fpm7.4 works, but line graph is there, with both php-fpm8.2 and php-fpm7.4 running

php -v
PHP 8.2.5 (cli) (built: Apr 14 2023 04:39:46) (NTS)
Copyright (c) The PHP Group
Zend Engine v4.2.5, Copyright (c) Zend Technologies
    with Zend OPcache v8.2.5, Copyright (c), by Zend Technologi

jeanmarc77, 2023-04-26:
what version does it show on your browser ? https://code.highcharts.com/highcharts.js and https://code.highcharts.com/highcharts-more.js
You sure you use latest index.php ? I have no glitches

halfagascan, 2023-04-26:
Highcharts JS v10.3.3 (2023-01-20)

 Highcharts JS v11.0.0 (2023-04-26)
copied the index.php from your metern link issue

jeanmarc77, 2023-04-26:
It should show Highcharts V11.. force that damn resfresh :p
Also use index for 123s https://github.com/jeanmarc77/123solar/blob/main/index.php

halfagascan, 2023-04-26:
Not sure what the hell is going on.
Started looking for a howto force update Highcharts, my duck foo is not working.
Both of the links to check the version now show Highcharts JS v11.0.0 (2023-04-26)
Still have the line graph.
thanks

jeanmarc77, 2023-04-26:
and you sure you took the latest https://github.com/jeanmarc77/123solar/blob/main/index.php ?

jeanmarc77, 2023-04-26:
You have several inverters ?

halfagascan, 2023-04-26:
Just one

halfagascan, 2023-04-26:
here a screen from android brave browser
[image]

halfagascan, 2023-04-26:
to add
win10 firefox 112.0.1 line graph shows, dash gauge is good
win 10 microsoft edge 112.0.1722.58  dash gauge shows lines, last 20 days good

soe135, 2023-04-28:
For the last two days I've been having the same damn problem with the distorted dashboard display as halfagascan. The problem occurs with all different browsers that I use (Firefox, Chrome, Edge), regardless of the operating system (Windows, Linux or Android). The problem seems to start synchronously with the new version of highcharts.js (Highcharts JS v11.0.0 (2023-04-26)).  Any suggestions on how to fix the problem?

halfagascan, 2023-04-29:
So, is this an upstream bug with highcharts.js? if so can you link the report?

jeanmarc77, 2023-04-29:
@halfagascan There is not issue with highcharts. 
did you use latest index (on git) and clear the cache ?

halfagascan, 2023-04-29:
solved thanks

closed 2023-04-29 by halfagascan