jeanmarc77/123solar #48: 123solar don't start after reboot

emka70, 2023-06-06:
Hi Jeanmarc,
my 123solar won't start (anymore) automatically after rebooting my RaspPi :-(

I've updated already to the newest version, 123solar.service-file is in /etc/systems/system and service is enabled.

When I reboot and do _systemctl status 123solar.service_, i'll get

> ● 123solar.service - 123Solar
>    Loaded: loaded (/etc/systemd/system/123solar.service; enabled; vendor preset: enabled)
>    Active: inactive (dead) since Tue 2023-06-06 19:15:15 CEST; 2min 51s ago
>   Process: 627 ExecStartPre=/bin/sleep 60 (code=exited, status=0/SUCCESS)
>   Process: 986 ExecStart=/usr/bin/curl http://localhost/123solar/scripts/boot123s.php (code=exited, status=0/SUCC
>  Main PID: 986 (code=exited, status=0/SUCCESS)
> 
> Jun 06 19:15:15 RPi4Wozi curl[986]: <html><head>
> Jun 06 19:15:15 RPi4Wozi curl[986]: <title>404 Not Found</title>
> Jun 06 19:15:15 RPi4Wozi curl[986]: </head><body>
> Jun 06 19:15:15 RPi4Wozi curl[986]: <h1>Not Found</h1>
> Jun 06 19:15:15 RPi4Wozi curl[986]: <p>The requested URL was not found on this server.</p>
> Jun 06 19:15:15 RPi4Wozi curl[986]: <hr>
> Jun 06 19:15:15 RPi4Wozi curl[986]: <address>Apache/2.4.38 (Raspbian) Server at localhost Port 80</address>
> Jun 06 19:15:15 RPi4Wozi curl[986]: </body></html>
> Jun 06 19:15:15 RPi4Wozi systemd[1]: 123solar.service: Succeeded.
> Jun 06 19:15:15 RPi4Wozi systemd[1]: Started 123Solar.

So it looks like 123solar has been started.

Going to my browser however, I'm guided directly to the admin-site and there's the button on top coloured in red and saying "off" ??!

I can turn then to "on" and it all works fine... til the next reboot...

Have you already had that problem ? And eventually any solution for me ??

Thanx,
Michael

jeanmarc77, 2023-06-07:
Hi,
It's look like curl can't reach localhost. Can you ping localhost ?
What contains /etc/hosts ? Is 123s directory located in localhost/123solar/ ?
BTW, use "Issues" for bugs and not help..

emka70, 2023-06-07:
Hi,
sry for abusing "Issues" - but how will I know if a problem's a bug - or just a help-case ?

my hosts shows
```
127.0.0.1       localhost
::1             localhost ip6-localhost ip6-loopback
ff02::1         ip6-allnodes
ff02::2         ip6-allrouters

127.0.1.1       RPi4Wozi
```
and my 123s is directly located in /var/www
I already tried the raspi's IP instead of "localhost" in 123solar.service - with same results :-(

jeanmarc77, 2023-06-07:
You've got a 404 error, it mean the page does not exist.
'ping localhost' from your logger.
Also returns output of 'curl http://localhost/123solar/scripts/boot123s.php'

emka70, 2023-06-07:
Hi,I‘ll give it a try tomorrow!Thx a lotGesendet mit der mobilen Mail AppAm 07.06.23 um 18:27 schrieb Jean-Marc Louviaux
                
            
            
                
                    Von: "Jean-Marc Louviaux" ***@***.***>Datum: 7. Juni 2023An: "jeanmarc77/123solar" ***@***.***>Cc: "Author" ***@***.***>,"Kauli" ***@***.***>Betreff: Re: [jeanmarc77/123solar] 123solar don't start after reboot (Issue #48)
                
                
You've got a 404 error, it mean the page does not exist.
'ping localhost' from your logger.
Also returns output of 'curl http://localhost/123solar/scripts/boot123s.php'

jeanmarc77, 2023-06-08:
ok 123solar directory cannot be installed in base-directory or it won't allow updating

closed 2023-06-08 by jeanmarc77

emka70, 2023-06-08:
so thats the reason why i have always to update manually ??!
great - another point for changing my apache-conf ...