jeanmarc77/123solar #49: high load on raspberry

basti122303, 2023-06-09:
Hello,
I run 123solar on a raspberry 2 Mod. B with nginx and php-fpm.

` PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND `                                                                                                                                                            
`  634 www-data  20   0    7768   5900   4748 S  18.1   0.6   0:09.89 nginx`

In the nginx log I every secound somethink like:
`[09/Jun/2023:11:30:01 +0200] "GET /solar/programs/programlive.php?invtnum=1 HTTP/1.0"`

Why does it poll so often?
what is it for?
and how can i do it more efficient?

Best regards

jeanmarc77, 2023-06-09:
To show 'live' data you need to retrieve them..
If you don't show 123s pages, there is no programlive requests

basti122303, 2023-06-09:
Thanks for the fast answer.
Where can I set the interval?
Whats about websockets to do this?

jeanmarc77, 2023-06-09:
[line 387](https://github.com/jeanmarc77/123solar/blob/main/index.php) 
I won't change the index behavior, you can set pooling in dashboard.
If you find a better way to improve perf, tell me how

closed 2023-06-09 by jeanmarc77