jeanmarc77/123solar #50: Maybe a better systemd service unit

basti122303, 2023-06-15:
I had the problem that after a upgrade of serveral php-packages the 123solar daemon was not started properly.
Paths and depensies should be edited.

`# /etc/systemd/system/123solar.service`
`[Unit]`
`Description=123Solar`
`Requires=network.target`
`After=network.target nginx.service php-fpm.service`
`StartLimitIntervalSec=60`
`StartLimitBurst=5`
` `
`[Service]`
`Type=simple`
`WorkingDirectory=/var/www/html/solar/scripts/`
`ExecStart=/usr/bin/php 123solar.php`
`# force PIDFILE see 'man systemd.service'`
`ExecStartPost=/bin/sh -c 'systemctl show -p MainPID 123solar.service | grep -o "[[:digit:]]*" > /var/www/html/solar/scripts/123solar.pid'`
`ExecStopPost=/usr/bin/rm -f /var/www/html/solar/scripts/123solar.pid`
`Restart=always`
`RestartSec=2s`
` `
`[Install]`
`WantedBy=default.target`

jeanmarc77, 2023-06-15:
Hi,
Thanks for the tips, I shall give it a try.
The 123solar.pid is make via boot123s.php or the interface, if do via systemctl can i trun off and on via the browser ?

basti122303, 2023-06-15:
I have not test if you can turn it on or off via browser.
But you can see the "state" of the service.
I think it is more confusing when the Service unit active and runing, when the admin page show the red button.

jeanmarc77, 2023-06-16:
Thanks, it have been pushed in git. All work fine except you don't get the log Starting/Stopping in 123s events.txt

basti122303, 2023-06-16:
> Maybe we could find a trick to chmod the .pid file to allow running either systemd or the interface ?

See https://github.com/jeanmarc77/123solar/pull/51

jeanmarc77, 2023-06-16:
it's ok your systemd file run as User=www-data/Group=www-data

basti122303, 2023-07-11:
```
root@c2:~# sudo -u www-data -s /bin/bash

www-data@c2:/root$ systemctl status 123solar.service
● 123solar.service - 123Solar
     Loaded: loaded (/etc/systemd/system/123solar.service; enabled; preset: enabled)
     Active: active (running) since Fri 2023-07-07 14:09:08 CEST; 3 days ago
    Process: 32543 ExecStartPost=/bin/sh -c systemctl show -p MainPID --value 123solar.service  > /home/www/solar/scripts/123solar.pid (code=exited, status=0/SUCCESS)
   Main PID: 32542 (php)
      Tasks: 1 (limit: 2110)
     Memory: 5.5M
        CPU: 58min 41.056s
     CGroup: /system.slice/123solar.service
             └─32542 /usr/bin/php 123solar.php
www-data@c2:/root$

www-data@c2:/root$ sudo systemctl stop 123solar.service
www-data@c2:/root$

www-data@c2:/root$ systemctl status 123solar.service
○ 123solar.service - 123Solar
     Loaded: loaded (/etc/systemd/system/123solar.service; enabled; preset: enabled)
     Active: inactive (dead) since Tue 2023-07-11 13:10:12 CEST; 12s ago
   Duration: 3d 23h 1min 4.614s
    Process: 32542 ExecStart=/usr/bin/php 123solar.php (code=killed, signal=TERM)
    Process: 32543 ExecStartPost=/bin/sh -c systemctl show -p MainPID --value 123solar.service  > /home/www/solar/scripts/123solar.pid (code=exited, status=0/SUCCESS)
    Process: 13057 ExecStopPost=/usr/bin/rm -f /home/www/solar/scripts/123solar.pid (code=exited, status=0/SUCCESS)
   Main PID: 32542 (code=killed, signal=TERM)
        CPU: 58min 41.235s
www-data@c2:/root$

www-data@c2:/root$ sudo systemctl start 123solar.service
www-data@c2:/root$

www-data@c2:/root$ systemctl status 123solar.service
● 123solar.service - 123Solar
     Loaded: loaded (/etc/systemd/system/123solar.service; enabled; preset: enabled)
     Active: active (running) since Tue 2023-07-11 13:10:35 CEST; 2s ago
    Process: 13067 ExecStartPost=/bin/sh -c systemctl show -p MainPID --value 123solar.service  > /home/www/solar/scripts/123solar.pid (code=exited, status=0/SUCCESS)
   Main PID: 13066 (php)
      Tasks: 1 (limit: 2110)
     Memory: 5.2M
        CPU: 453ms
     CGroup: /system.slice/123solar.service
             └─13066 /usr/bin/php 123solar.php
www-data@c2:/root$

flanesi, 2023-08-03:
I did a fresh installation of 123Solar on a Raspberry running Raspberry Pi OS Bullseye (Debian 11) with apache and PHP 8.2.8 (no php-fpm installed).
With the old service (123solar_old.service), 123solar starts at boot but cannot be stopped from the browser.

With the new service as below:
```
[Unit]
Description=123Solar
Requires=network.target
After=network.target apache2.service
StartLimitIntervalSec=60
StartLimitBurst=5
 
[Service]
Type=simple
WorkingDirectory=/var/www/123solar/scripts/
ExecStart=/usr/bin/php 123solar.php
# force PIDFILE see 'man systemd.service'
ExecStartPost=/bin/sh -c 'systemctl show -p MainPID --value 123solar.service > /var/www/123solar/scripts/123solar.pid'
ExecStopPost=/usr/bin/rm -f /var/www/123solar/scripts/123solar.pid
User=www-data
Group=www-data
Restart=always
RestartSec=30s
#Since version 7.4 php-fpm is hardened
ReadWritePaths = /var/lock
 
[Install]
WantedBy=default.target
```

and the file /etc/sudoers.d/www-data
```
# This file should placed in /etc/sudoers.d/ as www-data or something else
# Username should be edited for your needs
www-data ALL=(root) NOPASSWD: /usr/bin/systemctl stop 123solar.service
www-data ALL=(root) NOPASSWD: /usr/bin/systemctl start 123solar.service
www-data ALL=(root) NOPASSWD: /usr/bin/systemctl is-enabled 123solar.service
```
with owner root and permissions 0440,
123Solar starts regularly on boot and the service is active:
```
pi@raspberrypi:~ $ systemctl is-enabled 123solar.service
enabled
```
```
sudo systemctl status 123solar.service

● 123solar.service - 123Solar
     Loaded: loaded (/etc/systemd/system/123solar.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2023-07-31 00:59:18 CEST; 12h ago
    Process: 646 ExecStartPost=/bin/sh -c systemctl show -p MainPID --value 123solar.service > /var/www/123solar/scripts/123solar.pid (code=exited, status=0>
   Main PID: 645 (php)
      Tasks: 3 (limit: 1595)
        CPU: 7min 29.368s
     CGroup: /system.slice/123solar.service
             ├─ 645 /usr/bin/php 123solar.php
             ├─5738 sh -c aurora -b -a 2 -c -T -Y3 -l3 -w10 -d0 -e /dev/ttyUSB0 2> /tmp/de.err
             └─5739 aurora -b -a 2 -c -T -Y3 -l3 -w10 -d0 -e /dev/ttyUSB0
```

Executing from the terminal:
`sudo systemctl stop 123solar.service`
123Solar stops regularly and with the command:
`sudo systemctl start 123solar.service`
123solar restarts and works, but if you query the service it seems to be inactive, even though 123solar works regularly.
```
pi@raspberrypi:~ $ sudo systemctl status 123solar.service
● 123solar.service - 123Solar
     Loaded: loaded (/etc/systemd/system/123solar.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Mon 2023-07-31 13:59:05 CEST; 57s ago
    Process: 16774 ExecStart=/usr/bin/php 123solar.php (code=killed, signal=TERM)
    Process: 16777 ExecStartPost=/bin/sh -c systemctl show -p MainPID --value 123solar.service > /var/www/123solar/scripts/123solar.pid (code=exited, status>
    Process: 17198 ExecStopPost=/usr/bin/rm -f /var/www/123solar/scripts/123solar.pid (code=exited, status=0/SUCCESS)
   Main PID: 16774 (code=killed, signal=TERM)
        CPU: 315ms
```
The behaviour is the same if you perform a first stop and a subsequent start from the browser.

Now it is no longer possible to stop 123Solar either from the browser or from the command line.
123Solar always remains active with green button.

Do you have any ideas to solve this problem?

basti122303, 2023-08-03:
Have a look at the permisson of the PID file.
I'am not shure if apache is run as www-data. So edit the User/Group in the service unit.
I think if the PID file exist's there is a green button in the browser.

flane71, 2023-08-03:
www-data is the user that Apache use by default in Raspbian.
the PID file has group and owner www-data and permissions 0644 when 123solar starts at boot and also after the first stop and start.

closed 2023-09-01 by basti122303