jeanmarc77/123solar #59: PHP8.2: Deprecated: Using ${expr} (variable variables) in strings is deprecated, use {${expr}} instead

sovking, 2023-09-23:
Recently I installed another instance of 123solar in a host where is running apache and php 8.2.
In php 8.2 the use of ${expr} is deprecated, and 123solar uses frequently this syntax. In my installation I corrected it, but it would be nice if this change will be implemented also in the main branch.

jeanmarc77, 2023-09-27:
Hi,
Thanks for notifying, i don't code much these times. How to change ${'var' . $i} for example ?

jeanmarc77, 2023-09-30:
{${'var'.$i}}

jeanmarc77, 2023-11-01:
ok with quote
if ("{${'var'.$i}}") {
echo 'ok';
}

speeskeek, 2024-01-09:
> Recently I installed another instance of 123solar in a host where is running apache and php 8.2. In php 8.2 the use of ${expr} is deprecated, and 123solar uses frequently this syntax. In my installation I corrected it, but it would be nice if this change will be implemented also in the main branch.

I dont know if I have the same issue? did your installation run on php 8.2? or did it stop?
can I get your (altered) files?

hmakmur, 2024-01-29:
> Recently I installed another instance of 123solar in a host where is running apache and php 8.2. In php 8.2 the use of ${expr} is deprecated, and 123solar uses frequently this syntax. In my installation I corrected it, but it would be nice if this change will be implemented also in the main branch.

Would you please share your code changes so others can use it? I have upgraded my Ubuntu20.04 to Ubuntu 22.04 and now my 123solar wont work because my php is now version 8.x.

hmakmur, 2024-01-30:
> Would you please share your code changes so others can use it? I have upgraded my Ubuntu20.04 to Ubuntu 22.04 and now my 123solar wont work because my php is now version 8.x.

Just for completeness,  I reinstalled my software to latest and it seems to work with php 8.1. 
-

jeanmarc77, 2024-03-05:
BTW i use PHP 8.3.2 and it runs.
If someone have the willing to change all variables as mentioned Nov 1, 2023 as i don't have much times to do so, that's would be kind

NevilleDavis, 2024-03-18:
Hi I decide to try and check update for this problem with the deprecated variable structure. What I have done is in php.ini enabled the error to be logged on the production value, that is you get on a single pass on the script the line numbers of the code that the error exists...then edit those as required. The text with error is appearing in the upper section of the displayed page, I make notes of the line numbers. When you again enter that page you do not see those errors (until a change in script or the next start up etc.
This is the section of php.ini
; Common Values:
;   E_ALL (Show all errors, warnings and notices including coding standards.)
;   E_ALL & ~E_NOTICE  (Show all errors, except for notices)
;   E_ALL & ~E_NOTICE & ~E_STRICT  (Show all errors, except for notices and coding standards warnings.)
;   E_COMPILE_ERROR|E_RECOVERABLE_ERROR|E_ERROR|E_CORE_ERROR  (Show only errors)
; Default Value: E_ALL
; Development Value: E_ALL
Production Value: E_ALL & ~E_DEPRECATED & ~E_STRICT
; https://php.net/error-reporting
error_reporting = E_ALL & ~E_DEPRECATED & ~E_STRICT
All this is time consuming but I have managed 123 now to be error free...now to do meterN.

This is for those trying to find a simple way of finding the line numbers....just so many lines :)

I have been using 123 since 2013 and meterN since 2014 I think still going strong ...thanks Jeanmarc

NevilleDavis, 2024-03-22:
It turned out to be easier than I thought by doing this method. I have both 123solar and meterN no longer having PHP report errors.