jeanmarc77/123solar #66: Sofar solar (Alias zucchetti) 1100tl-3300tl G1 rs485 modbus protocol implementation

guadax, 2024-03-09:
Hi, just a question: protocol php file for the above inverter is almost completed (when completed i can post it). The only issues i have is on the variable

$KWHT is the total counter (a Wh precision is needed)

well... modbus register give me only total produce KWH precision (not watt):

[image]

without watt/hours precision the main production page is empty (maybe i'm wrong).

Any suggestions?

jeanmarc77, 2024-03-09:
Hi, 
Yes it is mandatory. Maybe you can also use a daily counter more precise ?
If think mi600-webif.php use some sort of calculation to get a wh precision

guadax, 2024-03-09:
all the data available below.... so i've to replace the total production(Produzione cumulativa (kw)) with daily production(Produzione giorno (kw))?

[image]

jeanmarc77, 2024-03-09:
i don't think so, the daily counter don't seem to have a Wh precision

guadax, 2024-03-09:
Seems to work with daily production
[image]

jeanmarc77, 2024-03-11:
Yes i think so, avg power need to have Wh precison or you'll get bumps in the chart

guadax, 2024-03-11:
no way: adding watt/hour precision manually didn't solve the problem.... so is elsewhere.... could you tell me the php file with implements the avg plotting function?

guadax, 2024-03-11:
added random last number (watt precision) not equal to zero
[image]

[image]