jeanmarc77/123solar #70: 2024 not being populated

maurizioandreotti, 2024-04-23:
since the beginning of the year the system, although it reads data from the inverter:

[image]

it seems not showing nor storing the data correctly:
[image]

[image]

 it's not populating the 2024 data.

[image]

I tried downloading a fresh version from github and reinstall, but howthing changed.

what shall i do?

jeanmarc77, 2024-04-23:
Hi,
github code is work in progress, you need to releases https://github.com/jeanmarc77/123solar/releases
Check Files permissions 123solar/admin/fperms.php

maurizioandreotti, 2024-04-23:
[image]

jeanmarc77, 2024-04-23:
Weird.. check your webserver error log

maurizioandreotti, 2024-04-23:
thanks.

i see no issues shown in any of the log (ngnix, php etc). but trying to copy the 123solar data folder i am now getting read errors..... so i think i got disk issue,

will see to reinstall the raspberry

maurizioandreotti, 2024-04-23:
I checked and info page say that i am running 123Solar 1.8.4.5

in file 123solar.err i see thousands of these notices:
PHP Warning:  include(../config/pvoutput/consumption/.php): failed to open stream: No such file or directory in /var/www/html/123solar/scripts/123solar.php on line 459
PHP Warning:  include(): Failed opening '../config/pvoutput/consumption/.php' for inclusion (include_path='.:/usr/share/php') in /var/www/html/123solar/scripts/123solar.php on line 459
PHP Notice:  Undefined variable: PVOT1 in /var/www/html/123solar/scripts/123solar.php on line 461
PHP Notice:  Undefined variable: PVOT1 in /var/www/html/123solar/scripts/123solar.php on line 462
PHP Warning:  include(../config/pvoutput/temperature/.php): failed to open stream: No such file or directory in /var/www/html/123solar/scripts/123solar.php on line 462
PHP Warning:  include(): Failed opening '../config/pvoutput/temperature/.php' for inclusion (include_path='.:/usr/share/php') in /var/www/html/123solar/scripts/123solar.php on line 462
PHP Notice:  Undefined variable: APIKEY1 in /var/www/html/123solar/scripts/123solar.php on line 470
PHP Notice:  Undefined variable: SYSID1 in /var/www/html/123solar/scripts/123solar.php on line 471
PHP Notice:  Undefined variable: PVOEXT1 in /var/www/html/123solar/scripts/123solar.php on line 480
PHP Notice:  Undefined variable: SYSID1 in /var/www/html/123solar/scripts/123solar.php on line 490
PHP Notice:  Undefined variable: PVOEXT1 in /var/www/html/123solar/scripts/123solar.php on line 491
PHP Notice:  Undefined variable: PVOUTPUT11 in /var/www/html/123solar/scripts/123solar.php on line 397
PHP Notice:  Undefined variable: PVOT1 in /var/www/html/123solar/scripts/123solar.php on line 451
PHP Notice:  Undefined variable: PVOC1 in /var/www/html/123solar/scripts/123solar.php on line 458
PHP Notice:  Undefined variable: PVOC1 in /var/www/html/123solar/scripts/123solar.php on line 459

jeanmarc77, 2024-04-23:
Did you set "Consumption" 123solar/admin/admin_pvo.php ?
What show config/config_pvoutput.php ? (obfuscate values!)

maurizioandreotti, 2024-04-23:
no, it was always set to no:

[image]

jeanmarc77, 2024-04-23:
and the content of config/config_pvoutput.php ?

maurizioandreotti, 2024-04-23:
maurizio@aurora:/var/www/html/123solar $ more config/config_pvoutput.php
<?php
if(!defined('checkaccess')){die('Direct access not permitted');}

// ### PVOUTPUT.org
$NUMPVO=0;

$SYSID1='0';
$APIKEY1='';
$PVOC1='no';
$PVOT1='inverter';
$PVOEXT1=false;
$PVOUTPUT11=false;

$cfgver=1451373026;
?>

jeanmarc77, 2024-04-23:
it's empty, so save PVoutput cfg

maurizioandreotti, 2024-04-23:
anyhow my problem is tghat it soedn create the 2024 csv.

yesterday i was ssindg daily watts, and 2024 in the historic window, but then when the day changed it went badk to show only 2023.

[image]

jeanmarc77, 2024-04-23:
Live values are ephemerous and they are not recorded in csv files. Hence why i suppose it's a permission issue or maybe a faulty/full drive

maurizioandreotti, 2024-04-28:
Hi Juean,
it was a faulty SD card in the raspberry: at some point in time it deceided to be only readonly, so the raspy was booting adn working, but not saving any file at all.

I had around another one which turned out to be faulty (this time showing random /w errors), so i  had to get a brand new one. and reinstall everything from scratch.

at this point in time it seems working fine since ithe last 2 days, 

 I have 2024 data (well yes, i had to donwload the hisory from the aruora inveerter and setup the 2024.csv file for the past months)

the only strange thing which is see is that the "last 20 days" graph shows a production of 0,

[image]

 while, if I click on it i can see the graph of the last day (as weel as the day before, going further back i cannot see, but that is expected as the systsem was being reinstalled):
[image]

jeanmarc77, 2024-04-29:
great, you'll get previous days next

closed 2024-04-29 by jeanmarc77