jeanmarc77/123solar #72: Shelly EM protocol - IP field enable

flanesi, 2024-06-28:
I've set up a protocol for **[Shelly EM](https://www.shelly.com/en/products/shop/shelly-em-120a)**, and it seems to work fine. I attach it.

From the admin page I have the "IP or RS-485" and "Port" fields disabled and I used the comoption field for the IP address. (see attached screenshot)
How can I enable these fields?

Naturally, any suggestion to improve the protocol is welcome  :-)
Thanks

[Shelly EM - 123Solar protocol.zip](https://github.com/user-attachments/files/16026124/Shelly.EM.-.123Solar.protocol.zip)

[image]

jeanmarc77, 2024-06-28:
Hi Flanesi,
Thanks. You may edit admin/admin_invt.php "Allow fields" to enable port and adresses fields.
Can you push your code to git ?
cheers

flane71, 2024-06-28:
ok, I'll do some more tests and then publish it.
I update you. Thank you

flanesi, 2024-07-01:
The script seems to work well.
Every now and then he does a double recording of the data (see 6:25)

```
Time,I1V,I1A,I1P,I2V,I2A,I2P,I3V,I3A,I3P,I4V,I4A,I4P,G1V,G1A,G1P,G2V,G2A,G2P,G3V,G3A,G3P,FRQ,EFF,INVT,BOOT,SR,KWHT
05:25:00,,,,,,,,,,,,,230.4,0,0,,,,,,,0,0,,,,722.633
05:30:00,,,,,,,,,,,,,231.6,0,0,,,,,,,0,0,,,,722.633
05:35:00,,,,,,,,,,,,,231.5,0,0,,,,,,,0,0,,,,722.633
05:40:00,,,,,,,,,,,,,230.4,0,0,,,,,,,0,0,,,,722.633
05:45:00,,,,,,,,,,,,,230.1,0,0,,,,,,,0,0,,,,722.633
05:50:00,,,,,,,,,,,,,231.1,0,0,,,,,,,0,0,,,,722.633
05:55:00,,,,,,,,,,,,,230.8,0.01,2,,,,,,,0,0,,,,722.633
06:00:00,,,,,,,,,,,,,230.6,0,0,,,,,,,0,0,,,,722.633
06:05:00,,,,,,,,,,,,,228.5,0.02,4,,,,,,,0,0,,,,722.633
06:10:00,,,,,,,,,,,,,230.6,0.05,12,,,,,,,0,0,,,,722.633
06:15:00,,,,,,,,,,,,,230.4,0.09,20,,,,,,,0,0,,,,722.634
06:20:00,,,,,,,,,,,,,230.2,0.1,23,,,,,,,0,0,,,,722.636
06:25:00,,,,,,,,,,,,,229,0.08,18,,,,,,,0,0,,,,722.638
06:25:31,,,,,,,,,,,,,229,0.08,19,,,,,,,0,0,,,,722.638
06:30:00,,,,,,,,,,,,,228.9,0.11,25,,,,,,,0,0,,,,722.639
06:35:00,,,,,,,,,,,,,230.9,0.19,45,,,,,,,0,0,,,,722.642
06:40:00,,,,,,,,,,,,,231.2,0.29,68,,,,,,,0,0,,,,722.646
06:45:00,,,,,,,,,,,,,230.2,0.48,111,,,,,,,0,0,,,,722.652

```

What could be the cause?

brodikAE, 2025-01-07:
Hi flanesi, yesterday I also wrote the shellyem protocol not knowing that it had already been done by you.
If you want in my script there is the possibility to set whether the channel from which it takes the data reads positive (+) or negative (-) by setting a parameter in the COMOPTION.
It is necessary to modify the "admin/admin_invt.php" file to enable the IP address field.

flanesi, 2025-01-07:
Hi brodikAE, 
can you share your protocol?
Do you also get double entries sometimes?

brodikAE, 2025-01-18:
I haven't tried it out thoroughly yet as I have it set up in case reading data from the inverter is not possible (for example in case the RS485-USB converter fails).
In the "/123solar/admin/admin_invt.php" file the _$adresslist_ variable must be changed to the following:
`$adresslist = array(
	'aurora',
	'485solar-get',
	'SBFspot',
	'rklogger',
	'fronius-fslurp',
	'sdm120c',
	'sdm120c-pool',
	'fronius',
	'kaco',
	'piko',
	'shellyem',
	'solarlog',
	'abbuno'
);`
Put the files in "/123solar/scripts/protocols/"
[shellyem_B88.zip](https://github.com/user-attachments/files/18465393/shellyem_B88.zip)