jeanmarc77/123solar #80: Reading Data from Sunny Tripower 8.0 SE

AKlemm88, 2025-02-10:
Hello, I want to read Data from a Sunny Tripower 8.0 SE.

If I Click Test Connection there is an error:

timeout --kill-after=15s 10s /usr/local/bin/sbfspot.3/SBFspot -finq -q -123s=DATA -cfgSBFspot_1.cfg return : 00464E09 28 Mon Feb 10 21:37:38 2025 '' 1 1 1 1

error : /

I wrote it at the SBFspot Repo, and the Problem is the -q Parameter.

The reply in the SBFspot repo is the following:

"The -q parameter on the command line hides the output - except for the Grid Voltage Phase apparently. This is a bug and might influence 123solar if you are using this application.

Replace the -q with -v5 to see verbose output"

With this change, i can see some Data. But I cant use it in 123solar. Is there and workarround ore Solution?

Thank you
Best regards
Alex

jeanmarc77, 2025-02-11:
Hi Alex,

The answer [is here](https://github.com/jeanmarc77/123solar/blob/main/scripts/protocols/SBFspot.php) 

with v5 the output is really verbose and does not respect the 123s data frame hence why you will had no data.
the debug mode is used for "Test Connection" in the interface and also for 123s in debug mode.

I thing the debug output of SBFspot shall be redirected to 2>

jeanmarc77, 2025-02-11:
the stderr output (see https://www.howtogeek.com/435903/what-are-stdin-stdout-and-stderr-on-linux/)

AKlemm88, 2025-02-15:
I posted again in the SBFspot repository. The issue is that SBFspot ignores battery inverters when exporting to 123solar. SBF is trying to fix it for the next release.

closed 2025-02-15 by jeanmarc77